IEAWindSystems/IEA-15-240-RWT #141: Problems with running OpenFAST 3.4.1 with IEA 15MW Umain semisub version 1.1.4

maucollu, 2023-05-18:
## Description
I wanted to use the most updated version of OpenFAST compatible with the new released IEA 15MW UMaine semisubmerible model.
I followed the steps below, and when running the simulation for the default 10 seconds all is ok, but when augmenting the simulation time to 200 seconds, it runs into an instability that, at first, gives unrealistic oscillations in power (and other response variables), which forces the BEM model to be shut down, and then incurs into a FATAL ERROR around 150s into the simulation.

Not sure if it is a problem of controller (I'm not an expert in control systems) or aerodynamics.

I'm using openfast_x64.exe. on a Windows machine.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Download IEA 15MW from the github repository (v.1.1.4): [Link](https://github.com/IEAWindTask37/IEA-15-240-RWT.git)
2. Download OpenFAST 3.4.1 (last compatible) executable from github repository: [Link](https://github.com/OpenFAST/openfast/releases)
3.  Download `libdiscon.dll` from ROSCO controller (v2.7.0) library: [Link](https://github.com/NREL/ROSCO/tags)
4. Move the file `libdiscon.dll` to the folder where ServoDyn.dat is
5. Open ...ServoDyn.dat, and at line 77, substitute: `"/usr/share/miniconda3/envs/test/lib/libdiscon.so"` with `"libdiscon.dll"`
6. In ` IEA-15-240-RWT-UMaineSemi.fst` in the folder `OpenFAST\IEA-15-240-RWT-UMaineSemi\`, change `TMax` to 200
7. Run `openfast_x64.exe  IEA-15-240-RWT-UMaineSemi.fst`, in the folder `OpenFAST\IEA-15-240-RWT-UMaineSemi\`

### Current behavior

Generator speed:    0.7 RPM, Pitch angle:  87.6 deg, Power:    -0.0 kW, Est. wind Speed:   4.9 m/s
Generator speed:    0.6 RPM, Pitch angle:  90.0 deg, Power:    -0.0 kW, Est. wind Speed:   5.4 m/s
Generator speed:    1.1 RPM, Pitch angle:  90.0 deg, Power:    -0.0 kW, Est. wind Speed:   5.3 m/s
Generator speed:    0.5 RPM, Pitch angle:  90.0 deg, Power:    -0.0 kW, Est. wind Speed:   5.3 m/s
 Time: 150 of 200 seconds.  Estimated final completion at 16:55:13.

FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 46, blade 3):UA_CalcOutput:Mach number exceeds 1.0. Equations cannot be evaluated.

 OpenFAST encountered an error at simulation time 150.9 of 200 seconds.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

### Expected behavior

I woudl expect an initial transient followed by a normal regime situation

### Code versions
OpenFAST 3.4.1
IEA 15MW Umaine 1.1.4
ROSCO libdiscon.dll 2.7.0

dzalkind, 2023-05-18:
Hi, we have not observed this issue before.  Can you compress all your inputs into a zip file and share that here, so we can reproduce your issue and compare with our inputs?

dzalkind, 2023-05-18:
I have traced the issue to the ElastoDyn tower.  Using an older model seems to resolve the instability: https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/0b0cf299c1dcf4e85d520a1029d64af548c1e5a7/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ElastoDyn_tower.dat

We will update the repo in the coming days.

maucollu, 2023-05-19:
Thanks a lot, @dzalkind  - now it seems to work well!
Just out of curiosity, what was wrong with the old elastodyn tower.dat?

gbarter, 2023-05-19:
The floating simulation becomes numerically unstable with a tower damping value of 0.32%, which is the value used in the monopile model and a value that has been recommended to us by industry.  We have reverted the damping value back to 1% in the upcoming v1.1.5 release and added some comments to the [FAQ Wiki entry](https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#what-are-the-values-of-structural-damping).

jennirinker, 2023-05-22:
Is that damping percentage log dec or percent critical? I'm working on the monopile tower this week and can make sure the UMaine tower is consistent while I'm at it

ptrbortolotti, 2023-05-22:
that's percent critical

gbarter, 2023-05-22:
@jennirinker - Take a look at #142 .  I tried to revert to some older values, but that doesn't guarantee consistency between HAWC2 and OF.  Let me know if values should change (or just commit to the feature branch directly).

ALBERTUTRERA, 2024-04-01:
Hi everyone,

I am currently using the 15 MW IEA monopile to run simulations for different wind speeds. Right now i am using inflowwind to simulate the turbine for 3 m/s. Given this situation, i have added a folder called 3ms and inside this folder there is the simulation file (.fst) and the inflowwind file. I do not understand why when running the simulation, at the specific time 23,865, the OpenFAST aborts. It seems AeroDYN is the problem but i am not able to trace the issue. This is the part that appears in the command window:

FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 50, blade 3):UA_CalcOutput:Mach number exceeds 1.0. Equations cannot be evaluated.

 OpenFAST encountered an error at simulation time 23.865 of 100 seconds.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

I attach my folders HERE (https://upm365-my.sharepoint.com/:f:/g/personal/a_utrera_alumnos_upm_es/Ek0am3r3uqxEkdXNWnodhhIBke0-YnzSODhA3IFRfwxqsw?e=02qtKd)  in order for you to see if you detect where is the problem. My simulation is in the 3ms folder because the first wind type that i want to test is for 3 m/s. I am really grateful if you could help me.

Best regards

ptrbortolotti, 2024-04-01:
Is the 3 m/s simulation the only one stopping? If so, check the rotor speed, the rotor might simply be shutting down because you're below cut in wind speed. At very low rpm BEM might simply stop working. Let us know if this is the case

ALBERTUTRERA, 2024-04-01:
I am only running for the 3m/s yeah! For the rest i have not tried yet!

I did The simulation again and The rotor speed output values are negative from 17 s to 23 seconds . IS this what you refer? I mean , is the negative value the cause?

BEST regards

ptrbortolotti, 2024-04-01:
negative rpm won't make BEM happy :)
try the larger wind speeds please
I also believe that you're using a single ElastoDyn main file. If that's the case, you're using a single initial condition in terms of pitch, rpm, deflections. That's probably not good enough for your initial transients

ALBERTUTRERA, 2024-04-01:
I have tried the simulation for wind speeds (12m/s and 15 m/s), and there are still negative value for rotor speed. How can i make the simulation to run properly regardless of the wind speed(taking into account that is in the interval between the cut-in wind speed and cut-out wind speed)?

Apart from this, i have tried the simulation for 21 m/s and the following message is shown in my screen:

** WARNING: Tower strike. **  This warning will not be repeated though the condition may persist.

FAST_Solution:FAST_AdvanceStates:AD_UpdateStates:SetDisturbedInflow:TwrInfl:Tower strike.

 OpenFAST encountered an error at simulation time 15.335 of 100 seconds.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

Thanks for helping

ptrbortolotti, 2024-04-01:
have you been adjusting your initial condition? The Rotor Performance tab in this excel can be of help https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/Documentation/IEA-15-240-RWT_tabular.xlsx

ALBERTUTRERA, 2024-04-01:
I have checked out the excel, i updated the rotor speed initial condition for 5 m/s. Regarding the blade initial pitch, i have not found anything in the excel...

The tower strike announcement is related to the tower displacement initial condition?

ptrbortolotti, 2024-04-01:
pitch angle is in column B. tower strike is probably caused by the pitch angle being badly off at the start

ALBERTUTRERA, 2024-04-01:
Thank you for the pitch answer.
For the parameters NacYaw, TTDspFA, TTDspSS, OoPDefl and IPDefl i have not found anything neither in the excel nor in other documentations. I suppose TTDspFA, TTDspSS can be set as 0, but for the rest i am not sure.

I have changed the blade pitch and there is still a fatal error due to the tower strike.

ptrbortolotti, 2024-04-01:
you can try to add a couple of degrees to your initial pitch. this should slow down the initial response of the rotor. if this doesn't work, send me the files and I can take a quick look

ALBERTUTRERA, 2024-04-01:
Here i leave the files: https://upm365-my.sharepoint.com/:f:/g/personal/a_utrera_alumnos_upm_es/Ekbiet-GcxRBoLWnFRt0pu0B9PffgF41ML1p09fY0KRKgQ?e=0E877g. I have been running the simulation in the folder 15ms

ptrbortolotti, 2024-04-01:
thank you. the problem is that you are not calling the modules hydrodyn and subdyn in your fst file. however you have the platform DOFs turned on in elastodyn

closed 2024-07-11 by gbarter