IEAWindSystems/IEA-15-240-RWT #146: A request of WAMIT result with COG = (0,0,-14.4)

hachikoi1, 2023-05-28:
# Description of feature
Describe the feature here and provide some context. Under what scenario would this be useful?
Dear developers
First of all thank you very much for your work. I am working on the hydrodynamic load calculation of UMaine via WEC-Sim developed by NREL, and compare the difference between WEC-Sim and OpenFAST-HydroDyn. Unlike PRP (0,0,0) in OpenFAST, WEC-Sim calculates hydrodynamic loads based on the platform COG (0,0,-14.4). Unfortunately, I don't have a WAMIT license. Could you please calculate the hydrodynamic loads at the COG using WAMIT and share the results? I will be happy to share the WEC-Sim input files in this post at a later date.
To end, thank you very much for your contribution!
Best regards,

# Potential solution
Can you think of ways to implement this?
I think it is only necessary to replace  **0      0      0      XCG** with **0      0      -14.4      XCG** in the fourth line of IEA-15-240-RWT-master\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\HydroData\wamit_inputs_1stOrder\IEA-15-240-RWT.frc

In addition, OpenFAST requires the correct COG coordinates to be specified in IEA-15-240-RWT-master\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\HydroData\wamit_inputs_2ndOrder\IEA-15-240-RWT_2nd.frc, but I found **0	0	-2.32	XCG** in the fourth line of IEA-15-240-RWT_2nd.frc. Shouldn't it be **0	0	-14.4	XCG**?

mattEhall, 2023-05-30:
Hi, have you considered just transforming the existing WAMIT data files that are provided in the repo? If I'm thinking about this correctly, the math to convert the data from the current reference point at z=0 to a reference point with z=-14.4 could be implemented in a script -- probably much more quickly than it would take to recompute all the data files with WAMIT. The entries that would need to be adjusted would be the roll and pitch DOFs, and any cross-coupling (off-diagonal) entries involving those DOFs.

luwang00, 2023-05-30:
Hi @hachikoi1 

I agree with @mattEhall that it's probably better to transform the existing WAMIT data about z=0 to about z=-14.4 m. Make sure you account for the fact that both the moments and the rotation (roll and pitch) are about a new reference point.

Also, it is necessary to approximate the first-order motion response of the whole system when computing the second-order wave excitation. That's why XCG=0,0,-2.32m in the second-order input file. This XCG is that of the whole system including the tower and the turbine, not just the platform.

hachikoi1, 2023-05-30:
Dear [mattEhall](https://github.com/mattEhall)
Thank you very much for your advice. If I understand correctly, you mean that the hydrodynamic coefficients at COG (0 0 -14.4) can be evaluated based on those at PRP (0 0 0) using coordinate conversion, similar to the method mentioned in https://github.com/WEC-Sim/WEC-Sim/issues/1051. The coordinate conversion method works for hydrostatic stiffness matrix, but may give rise to unstable results for radiation and wave excitation terms, i.e. the results of the conversion vary considerably at each time step.
Or you mean that the conversion of the reference point of hydrodynamic coefficients can be achieved by multiplying the hydrodynamic coefficient matrix by conversion matrix?

Best regards,

hachikoi1, 2023-05-30:
Dear [mattEhall](https://github.com/mattEhall) and [luwang00](https://github.com/luwang00)
Thank you very much for your suggestion. I will try the conversion method you described.

Best regards,

closed 2023-05-30 by hachikoi1

hachikoi1, 2023-05-31:
> Dear [mattEhall](https://github.com/mattEhall) and [luwang00](https://github.com/luwang00) Thank you very much for your advice. If I understand correctly, you mean that the hydrodynamic coefficients at COG (0 0 -14.4) can be evaluated based on those at PRP (0 0 0) using coordinate conversion, similar to the method mentioned in [WEC-Sim/WEC-Sim#1051](https://github.com/WEC-Sim/WEC-Sim/issues/1051).
> 
> Take added mass term as an example. Theoretically, we can get added mass load and velocity at COG based on the added mass load, displacement and velocity at PRP. With the constraint conditions of Add_{COG}(1,1) = Add_{PRP}(1,1), Add_{COG}(2,2) = Add_{PRP}(2,2), Add_{COG}(3,3) = Add_{PRP}(3,3), Add_{COG}(6,6) = Add_{PRP}(6,6) and Add_{COG}(4,4) = Add_{COG}(5,5), we can ontain added mass matrix at COG, by solving the equation system of F_{addCOG} = -Add_{COG}*a_Cog at each time step, totalling 31 quantities. And multiple time steps must be introduced into the system of equations to solve them simultaneously
> 
> However, the method may give rise to unstable results for radiation and wave excitation terms, i.e. the results of the conversion vary considerably at each time step.
> 
> Or you mean that the conversion of the reference point of hydrodynamic coefficients can be achieved by multiplying the hydrodynamic coefficient matrix by conversion matrix?
> 
> Best regards,

mattEhall, 2023-05-31:
A coordinate conversion is what I had in mind. I'm not quickly grasping what you wrote or what is in the WEC-Sim link. That said, the link looked to be about transforming the quadratic drag matrix. For what it's worth, that would be very difficult because that matrix is for nonlinear terms, so I don't think it would be possible to accurately shift its reference point.

This function here is an example of the sort of transformation I was thinking of (for matrices): https://github.com/WISDEM/RAFT/blob/8c0445086969f480e76b9409087cbd56dce19437/raft/helpers.py#L321
https://github.com/WISDEM/RAFT/blob/8c0445086969f480e76b9409087cbd56dce19437/raft/helpers.py#L321

It is simpler for excitation because you just adjust the pitch and roll moments by adding the surge or sway forces multiplied by the z shift in the reference point.

Matt

hachikoi1, 2023-06-01:
Dear [mattEhall](https://github.com/mattEhall)
Many thanks for your sharing.
Best regards,
Tianyuan