IEAWindSystems/IEA-15-240-RWT #151: Report typo in soil modulus of elasticity for monopile?

jennirinker, 2023-06-12:
## Description

Per Table 4-1 in the original report
[image]
the soil is listed as having a shear modulus of 140 MPa.

However, this is causing stability issues in the HAWC2 model, and I think it might be a typo. From [this reference](https://www.northernarchitecture.us/earthquake-engineering/shear-modulus.html), the range of the _Young's modulus_ for the soil of dense sand and gravel is from 90 - 200 MPa, not the shear modulus. A shear modulus of 140 MPa corresponds to a Young's modulus of 392 MPa, which is much, much too large.

So is the value in the table meant to be the Young's modulus, and not the shear modulus? If that's the case, then the soil shear modulus should be 50 MPa, not 140 MPa, is that correct?

gbarter, 2023-06-13:
@jennirinker - The reference in the report is an old one and I think the modeling method came from Rick Damiani many years ago.  The shear modulus is taken from Table 4-8 (pg 69 in the book / 83 in the pdf) by converting 20k psi to MPa.  The equivalent spring-stiffness is in Table 4-1 (pg 58/72)
[suresh-aryapincus-design-of-structures-and-foundation-for-vibrating-machinepdf.pdf](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/11729902/suresh-aryapincus-design-of-structures-and-foundation-for-vibrating-machinepdf.pdf)

I honestly don't know enough of the geotech fundamentals to square this reference with the site/paper you linked to.

On the question of the lateral stiffness- are you asking about the soil model?  The monopile has constant diameter & thickness below the mudline, so the side-side stiffness shouldn't change.

gbarter, 2023-06-19:
You rock, @RBergua !

jennirinker, 2023-06-19:
Excellent -- I also asked some of my industry contacts and they agree the value assumed in the report is reasonable.

My colleague pointed out a constraint issue I had with the monopile (model is unstable without a torsional constraint), so the numerical issue is now resolved.

Since we have established the shear modulus is a reasonable value for dense sand, I think it is fine to close this issue. We are using a simplistic (linear) soil model for the IEA 15 MW which we know is unrealistic, but I think it is fine to do so. We can do something more realistic for the 22 MW. :)

I will therefore close this issue, but @gbarter feel free to reopen if you think something about the model should be changed.

closed 2023-06-19 by jennirinker