IEAWindSystems/IEA-15-240-RWT #160: IEA-15-240-RWT-Umainesemi encountered bad real number

HANloading, 2023-09-29:
Hi，everyone:
   i encountered this error when running IEA-15-240-RWT-UMaineSemi

Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows).
Using legacy Bladed DLL interface.

FAST_InitializeAll:HydroDyn_Init:HydroDynInput_ProcessInitData: A requested output channel is
invalid

At line 78 of file D:\bld\rosco_1615353392809\work\src\ReadSetParameters.f90 (unit = 89, file = '.\IEA-15-240-RWT-UMaineSemi_DISCON.IN')
Fortran runtime error: Bad real number in item 2 of list input

the message At line 78 of file ReadSetParameters.f90 is       {  ! GenTemp = avrSWAP(1026)}

 i do not know that At line 78 of file ReadSetParameters.f90 have any problem.

is there someone who can help me to address this problem please?thanks a lot.

dzalkind, 2023-09-29:
Hi, this is a strange error message, indeed! In both ROSCO 2.7.0 and 2.8.0 on that line, no actual code exists to cause a failure.

Can you confirm which version of ROSCO you are using and if you're using [the DISCON file here](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_DISCON.IN) without changes?

In parallel, it's time to update the ROSCO inputs on this repo to 2.8.0.  I will do that, which may solve your issue.

HANloading, 2023-09-30:
> Hi, this is a strange error message, indeed! In both ROSCO 2.7.0 and 2.8.0 on that line, no actual code exists to cause a failure.
>
> Can you confirm which version of ROSCO you are using and if you're [...]

Thank you for your answer, I have solved this issue.But i also encountered some warning：

 Running HydroDyn.
  WARNING:  The requested output channel is invalid: WavesF1xi
  WARNING:  The requested output channel is invalid: WavesF1zi
  WARNING:  The requested output channel is invalid: WavesM1yi
  WARNING:  The requested output channel is invalid: WavesF2xi
  WARNING:  The requested output channel is invalid: WavesF2zi
  WARNING:  The requested output channel is invalid: WavesM2yi
  WARNING:  The requested output channel is invalid: WavesF2xi
  WARNING:  The requested output channel is invalid: WavesF2yi
  WARNING:  The requested output channel is invalid: WavesF2zi
  WARNING:  The requested output channel is invalid: WavesM2xi
  WARNING:  The requested output channel is invalid: WavesM2yi
  WARNING:  The requested output channel is invalid: WavesM2zi
 Generating incident wave kinematics and current time history.
 Calculating second order difference frequency wave kinematics.
 Calculating second order sum frequency wave kinematics.
 Reading in WAMIT output with root name ".\HydroData/IEA-15-240-RWT-UMaineSemi".
 Computing radiation impulse response functions and wave diffraction forces.
 Calculating second order difference-frequency force using the full quadratic transfer function.
Running MoorDyn (v2.0.0, 2022-12-08).
   This is MoorDyn v2, with significant input file changes from v1.
   Copyright: (C) 2022 National Renewable Energy Laboratory, (C) 2019 Matt Hall
   Parsing MoorDyn input file: .\IEA-15-240-RWT-UMaineSemi_MoorDyn.dat
Warning: invalid output specifier FX.  Must start with L, C, R, or B
Warning: invalid output specifier FY.  Must start with L, C, R, or B
Warning: invalid output specifier FZ.  Must start with L, C, R, or B
   Created mooring system:  3 lines, 6 points, 0 rods, 0 bodies.
   Finalizing initial conditions using dynamic relaxation.
   t=60  FairTen 1: 2.43287E+06, 2.44526E+06, 2.43537E+06
   Fairlead tensions did not converge within TMaxIC=60 seconds.
   MoorDyn initialization completed.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows).
Using legacy Bladed DLL interface.

FAST_InitializeAll:HydroDyn_Init:HydroDynInput_ProcessInitData: A requested output channel is
invalid

------------------------------------------------------------------------------
Running ROSCO-v2.8.0
A wind turbine controller framework for public use in the scientific field
Developed in collaboration: National Renewable Energy Laboratory
                            Delft University of Technology, The Netherlands
------------------------------------------------------------------------------
 ROSCO Warning: Did not find Echo in input file.  Using default value of            0
 ROSCO Warning: Did not find AWC_Mode in input file.  Using default value of            0
 ROSCO Warning: Did not find CC_Mode in input file.  Using default value of            0
 ROSCO Warning: Did not find StC_Mode in input file.  Using default value of            0
 ROSCO Warning: Did not find AWC_NumModes in input file.  Using default value of            0
 ROSCO Warning: Did not find correct size AWC_n in input file.  Using default value of [           0 ]
 ROSCO Warning: Did not find correct size AWC_harmonic in input file.  Using default value of [           0 ]
 ROSCO Warning: Did not find correct sizeAWC_freq in input file.  Using default value of [   0.0000000000000000      ]
 ROSCO Warning: Did not find correct sizeAWC_amp in input file.  Using default value of [   0.0000000000000000      ]
 ROSCO Warning: Did not find correct sizeAWC_clockangle in input file.  Using default value of [   0.0000000000000000      ]
 ROSCO Warning: Did not find CC_Group_N in input file.  Using default value of            0
 ROSCO Warning: Did not find correct size CC_GroupIndex in input file.  Using default value of [           0 ]
 ROSCO Warning: Did not find CC_ActTau in input file.  Using default value of    0.0000000000000000
 ROSCO Warning: Did not find StC_Group_N in input file.  Using default value of            0
 ROSCO Warning: Did not find correct size StC_GroupIndex in input file.  Using default value of [           0 ]
 ROSCO Warning: Did not find correct size Ind_CableControl in input file.  Using default value of [           0 ]
 ROSCO Warning: Did not find correct size Ind_StructControl in input file.  Using default value of [           0 ]
Generator speed:    7.6 RPM, Pitch angle:   1.0 deg, Power:     0.0 kW, Est. wind Speed:  10.0 m/s

FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 50, blade 2):UA_CalcOutput:Mach number exceeds 0.3. Theory is invalid. This warning will
not be repeated though the condition may persist.
Generator speed:    5.5 RPM, Pitch angle:   4.5 deg, Power: 10047.7 kW, Est. wind Speed:   8.6 m/s

i hope that you can help me again.
thanks a lot！

HANloading, 2023-10-03:
> This appears to be working for you, though it looks like you're using the ROSCO v2.8.0 dynamic library with an older input file. This setup works but assigns default values to the extra inputs.
> 
> The [latest DISCON input file](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/develop/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_DISCON.IN) is now in the develop branch and should have no warnings.
> 
> Best, Dan

Thank you so much！It worked.

closed 2024-07-11 by gbarter