IEAWindSystems/IEA-15-240-RWT #163: The VTK file can't be created when considering the difference-frequency 2nd-order wave kinematics?

Steven647, 2023-10-14:
Dear All,

When I was running IEA 15MW Umiane semi-submersible model with OpenFAST, the **WrVTK** in **IEA-15-240-RWT-UMaineSemi.fst** is switched to 2 and **VTK_type** to 1 , and then the following problem occured:

OpenFAST-v3.5.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 2021
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: May 17 2023
 - Time: 13:43:33
Execution Info:
 - Date: 10/14/2023
 - Time: 13:34:18+0800

OpenFAST input file heading:
    IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

Running ElastoDyn.
Running AeroDyn.
Running InflowWind.
Running HydroDyn.
  WaveTMax is less then the simulation time.  Wave data will repeat every WaveTMax seconds.
 Generating incident wave kinematics and current time history.
 Calculating second order difference frequency wave kinematics.

FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:1/VTK_fps is not an integer multiple of DT.
FAST will output VTK information at 13.333 fps, the closest rate possible.
FAST_InitializeAll:HydroDyn_Init: WaveElevSeries arrays for first and second order wave
elevations are of different sizes.

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

If the **WvDiffQTF** in **IEA-15-240-RWT-UMaineSemi_HydroDyn.dat** is setted to FALSE, the problem will disappear.

How to solve the above problems?

gbarter, 2023-10-16:
@luwang00 @andrew-platt Could you please respond here or wherever else the question is posted.

andrew-platt, 2023-10-16:
Please see the OpenFAST issue here: https://github.com/OpenFAST/openfast/issues/1809 and fix here: https://github.com/OpenFAST/openfast/pull/1621

Steven647, 2023-10-19:
> Please see the OpenFAST issue here: [OpenFAST/openfast#1809](https://github.com/OpenFAST/openfast/issues/1809) and fix here: [OpenFAST/openfast#1621](https://github.com/OpenFAST/openfast/pull/1621)

Thank you very much!!!

closed 2023-10-19 by Steven647