IEAWindSystems/IEA-15-240-RWT #165: The ServoDyn module seems doesn't work

DLUT-cabbage, 2023-10-19:
Hello:
    When I run version IEA-15-240-RWT-1.1.4 ([https://github.com/IEAWindTask37/IEA-15-240-RWT/releases/tag/v1.1.4] ) to calculate the aerodynamic response of the 15MW monopile wind turbine in steady wind Vhub=10.66m/s (Rated wind speed), the model runs fine, but the ServoDyn module seems doesn't work. The GenPwr and GenTq equal to 0 over time, and RtAreoCp and RtAreoCt drop to 0. It could be an oversight in my setup, but after trying, I still can't find the problematic slip-up. So, I'm uploading the model file now and look for help.

**Steps**
1. Download IEA 15MW from the github repository (v.1.1.4)
2. Download OpenFAST 3.4.1 (last compatible) executable from github repository
3. Download libdiscon.dll from ROSCO controller (v2.7.0) library and compile successfully
4. In "Inflowfile.dat", set the "WindType=1 (steady wind)", and set the horizontal windspeed to 10.66m/s
5. In"IEA-15-240-RWT-Monopile.fst", turn on the function switch "CompElast", "CompInflow", "CompAero", "CompServo", and setting the input file location.

**Behaviors**
[image]

**Model:** Aerodynamic calculation for IEA-15MW wind turbine under steady wind (AeroDyn+ElastoDyn+ServoDyn)
1.  Inflowfile
`------- InflowWind v3.01.* INPUT FILE -------------------------------------------------------------------------
IEA 15 MW Offshore Reference Turbine
---------------------------------------------------------------------------------------------------------------
False                  Echo        - Echo input data to <RootName>.ech (flag)
1                      WindType    - switch for wind file type (1=steady; 2=uniform; 3=binary TurbSim FF; 4=binary Bladed-style FF; 5=HAWC format; 6=User defined; 7=native Bladed FF)
0.0                    PropagationDir - Direction of wind propagation (meteoroligical rotation from aligned with X (positive rotates towards -Y) -- degrees)
0.0                    VFlowAng    - Upflow angle (degrees) (not used for native Bladed format WindType=7)
1                      NWindVel    - Number of points to output the wind velocity    (0 to 9)
0.0                    WindVxiList - List of coordinates in the inertial X direction (m)
0.0                    WindVyiList - List of coordinates in the inertial Y direction (m)
150.0                  WindVziList - List of coordinates in the inertial Z direction (m)
================== Parameters for Steady Wind Conditions [used only for WindType = 1] =========================
10.66                  HWindSpeed  - Horizontal windspeed                            (m/s)
150.0                  RefHt       - Reference height for horizontal wind speed      (m)
0.12                   PLexp       - Power law exponent                              (-)
================== Parameters for Uniform wind file   [used only for WindType = 2] ============================
"none"                 Filename_Uni - Filename of time series data for uniform wind field.      (-)
150.0                  RefHt_Uni   - Reference height for horizontal wind speed                (m)
240.0                  RefLength   - Reference length for linear horizontal and vertical sheer (-)
================== Parameters for Binary TurbSim Full-Field files   [used only for WindType = 3] ==============
"Kaimal.bts"                 FileName_BTS - Name of the Full field wind file to use (.bts)
================== Parameters for Binary Bladed-style Full-Field files   [used only for WindType = 4] =========
"none"                 FilenameRoot - Rootname of the full-field wind file to use (.wnd, .sum)
False                  TowerFile   - Have tower file (.twr) (flag)
================== Parameters for HAWC-format binary files  [Only used with WindType = 5] =====================
"none"                 FileName_u  - name of the file containing the u-component fluctuating wind (.bin)
"none"                 FileName_v  - name of the file containing the v-component fluctuating wind (.bin)
"none"                 FileName_w  - name of the file containing the w-component fluctuating wind (.bin)
64                     nx          - number of grids in the x direction (in the 3 files above) (-)
32                     ny          - number of grids in the y direction (in the 3 files above) (-)
32                     nz          - number of grids in the z direction (in the 3 files above) (-)
16.0                   dx          - distance (in meters) between points in the x direction    (m)
3.0                    dy          - distance (in meters) between points in the y direction    (m)
3.0                    dz          - distance (in meters) between points in the z direction    (m)
150.0                  RefHt_Hawc  - reference height; the height (in meters) of the vertical center of the grid (m)
-------------   Scaling parameters for turbulence   ---------------------------------------------------------
2                      ScaleMethod - Turbulence scaling method   [0 = none, 1 = direct scaling, 2 = calculate scaling factor based on a desired standard deviation]
1.0                    SFx         - Turbulence scaling factor for the x direction (-)   [ScaleMethod=1]
1.0                    SFy         - Turbulence scaling factor for the y direction (-)   [ScaleMethod=1]
1.0                    SFz         - Turbulence scaling factor for the z direction (-)   [ScaleMethod=1]
1.2                    SigmaFx     - Turbulence standard deviation to calculate scaling from in x direction (m/s)    [ScaleMethod=2]
0.8                    SigmaFy     - Turbulence standard deviation to calculate scaling from in y direction (m/s)    [ScaleMethod=2]
0.2                    SigmaFz     - Turbulence standard deviation to calculate scaling from in z direction (m/s)    [ScaleMethod=2]
-------------   Mean wind profile parameters (added to HAWC-format files)   ---------------------------------
12.0                   URef        - Mean u-component wind speed at the reference height (m/s)
2                      WindProfile - Wind profile type (0=constant;1=logarithmic,2=power law)
0.2                    PLExp_Hawc  - Power law exponent (-) (used for PL wind profile type only)
0.03                   Z0          - Surface roughness length (m) (used for LG wind profile type only)
0                      XOffset     - Initial offset in +x direction (shift of wind box) (-)
====================== OUTPUT ==================================================
False                  SumPrint    - Print summary data to <RootName>.IfW.sum (flag)
OutList      - The next line(s) contains a list of output parameters.  See OutListParameters.xlsx for a listing of available output channels, (-)
Wind1VelX
Wind1VelY
Wind1VelZ
[... 458 lines cut ...]
0.0                    TStart      - Time to begin tabular output (s)
3                      OutFileFmt  - Format for tabular (time-marching) output file (switch) {1: text file [<RootName>.out], 2: binary file [<RootName>.outb], 3: both}
True                   TabDelim    - Use tab delimiters in text tabular output file? (flag) {uses spaces if false}
"ES10.3E2"             OutFmt      - Format used for text tabular output, excluding the time channel.  Resulting field should be 10 characters. (quoted string)
---------------------- LINEARIZATION -------------------------------------------
False                  Linearize   - Linearization analysis (flag)
False                  CalcSteady  - Calculate a steady-state periodic operating point before linearization? [unused if Linearize=False] (flag)
3                      TrimCase    - Controller parameter to be trimmed {1:yaw; 2:torque; 3:pitch} [used only if CalcSteady=True] (-)
0.001                  TrimTol     - Tolerance for the rotational speed convergence [used only if CalcSteady=True] (-)
0.01                   TrimGain    - Proportional gain for the rotational speed error (>0) [used only if CalcSteady=True] (rad/(rad/s) for yaw or pitch; Nm/(rad/s) for torque)
0                      Twr_Kdmp    - Damping factor for the tower [used only if CalcSteady=True] (N/(m/s))
0                      Bld_Kdmp    - Damping factor for the blades [used only if CalcSteady=True] (N/(m/s))
2                      NLinTimes   - Number of times to linearize (-) [>=1] [unused if Linearize=False]
30.000000, 60.000000   LinTimes    - List of times at which to linearize (s) [1 to NLinTimes] [used only when Linearize=True and CalcSteady=False]
[... cut ...]
0                      WrVTK       - VTK visualization data output: (switch) {0=none; 1=initialization data only; 2=animation}
2                      VTK_type    - Type of VTK visualization data: (switch) {1=surfaces; 2=basic meshes (lines/points); 3=all meshes (debug)} [unused if WrVTK=0]
False                  VTK_fields  - Write mesh fields to VTK data files? (flag) {true/false} [unused if WrVTK=0]
15.0                   VTK_fps     - Frame rate for VTK output (frames per second){will use closest integer multiple of DT} [used only if WrVTK=2]`

    Looking forward to replying!

gbarter, 2023-10-22:
Is this still an issue?  I think you might need to be using the latest OpenFast (v3.5.1).  Your message suggets you are using 3.4.

DLUT-cabbage, 2023-10-24:
Hi @gbarter: 
    So grateful for your reply. As per your suggestion I updated the version of OpenFast (v3.5.1) and tried again but still not resolved. For testing, I did the following settings for _IEA-15-240-RWT-Monopile_ServoDyn.dat_: PCMode=0 and VSContrl=1, and the results are as follows. It looks like the generator will start if pitch control mode is turned off. So is it possible to troubleshoot the problem?
[image]

    I've uploaded the model for ease of communication. [https://github.com/DLUT-cabbage/Learning-Edition-Openfast-v3.5.1]. Anyway, thanks again for your reply and admiration for the work you and your team have done!

dzalkind, 2023-10-24:
Hi,

I'm not exactly sure what your target is or the issue.  I recommend using the models provided in this repo on the develop branch and changing them incrementally until they reach your desired target.  You'll also want to use ROSCO v2.8.0.

Best, Dan

closed 2024-07-11 by gbarter