IEAWindSystems/IEA-15-240-RWT #168: zsh: segmentation fault  openfast IEA-15-240-RWT-Monopile.fst

Ssurya013, 2023-10-20:
Hi,

I am new to OpenFAST platform. I am running OpeFAST from my Mac. I cloned the IEA-15-24-RWT develop branch. I changed the path for libdiscon.dylib in the servodyn file. While running the .fst file I get the following warnings and errors. May I ask how to resolve the warnings and errors, please?

(openfast_env) sb01@P6Q22Q IEA-15-240-RWT-Monopile % openfast IEA-15-240-RWT-Monopile.fst

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2023 National Renewable Energy Laboratory
 Copyright (C) 2023 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-HEAD-HASH-NOTFOUND
 Compile Info:
  - Compiler: GCC version 12.2.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: May 17 2023
  - Time: 10:30:13
 Execution Info:
  - Date: 10/20/2023
  - Time: 01:43:06+0100

 OpenFAST input file heading:
     IEA 15 MW offshore reference model monopile configuration

 Running ElastoDyn.
 Running AeroDyn.
 Running InflowWind.
 Running HydroDyn.
  Generating incident wave kinematics and current time history.
 Running SubDyn.
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 Warning: the SubDyn input file is not at the latest format!
          Visit: https://openfast.readthedocs.io/en/dev/source/user/api_change.html
 > Issue: Member table contains 6 columns instead of 7,  using default member directional cosines
 ID (-1) for all members. The directional cosines will be computed based on the member nodes for
 all members.
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
    Fixed bottom case detected
    Performing Craig-Bampton reduction 114 DOFs -> 0 modes + 6 DOFs
    Using static improvement method for gravity and ext. loads
    Calculating Full System Modes for output files
    Exporting Summary file
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers (using GNU Fortran for Linux).
 Using legacy Bladed DLL interface.

 FAST_InitializeAll:HydroDyn_Init:Waves_Init:VariousWaves_Init: The random number generator in use
 differs from the original code provided by NREL. This pRNG uses 8 seeds instead of the 2 in the
 HydroDyn input file.

  Time: 0 of 300 seconds.
                                                                               
------------------------------------------------------------------------------
Running ROSCO-v2.8.0
A wind turbine controller framework for public use in the scientific field    
Developed in collaboration: National Renewable Energy Laboratory              
                            Delft University of Technology, The Netherlands   
------------------------------------------------------------------------------
 ROSCO Warning: Did not find DT_Out in input file.  Using default value of    0.0000000000000000     
 ROSCO Warning: Did not find VS_ConstPower in input file.  Using default value of            0
 ROSCO Warning: Did not find PRC_Mode in input file.  Using default value of            0
 ROSCO:SetParameters:ParseInput_Int_Opt:Missing or default values are not allowed for F_NumNotchFilts. Please check control modes.

Program received signal SIGSEGV: Segmentation fault - invalid memory reference.

Backtrace for this error:
#0  0x102b2fcf7
#1  0x102b2ed23
#2  0x186df9a23
#3  0x1023a34af
zsh: segmentation fault  openfast IEA-15-240-RWT-Monopile.fst

gbarter, 2023-10-22:
Hello,

I just posted an update for ROSCO v2.8, so that might help with you error.  If not, we cannot anticipate all possible combinations of compilers and OS versions, so you might need to compile the ROSCO library yourself from [its source](https://github.com/nrel/rosco).

Cheers,
Garrett

Ssurya013, 2023-10-24:
Hi Garrett,

Thank you for the update. I was however getting the same set of warnings and errors even after the update . I then compiled the ROSCO library from its source, still it did not solve the issues. The 15MW UMaineSemi within ROSCO repository Test_Cases runs without any issues. I then compared the input files for both the cases and found out that there are a few variables those were missing in the input file compiled by ROSCO for the monopile case. After adding those variables manually, the monopile model now runs without any issues. Thank you very much for your time. Much appreciated.

Cheers,
Surya

gbarter, 2023-10-25:
Hi Surya,

I am glad to hear that things are working although I didn't fully follow your last comment.  If there are corrections that need to be posted to one of the repository files, please let me know.

Cheeres,
Garrett

closed 2024-07-11 by gbarter