IEAWindSystems/IEA-15-240-RWT #169: Xt: [1546.e+6, 0.0, 0.0]       Xc: [1047.e+6, 0.0, 0.0]       S: [0.0, 0.0, 0.0]

ncepulibei, 2023-10-25:
# Description of feature
Xt: [1546.e+6, 0.0, 0.0]       Xc: [1047.e+6, 0.0, 0.0]       S: [0.0, 0.0, 0.0]
I am conducting a failure analysis, but I found that some of the data for CarbonUD is 0 in the line of 817、818 and 819 IEA-15-240-RWT.yaml file.  Is this normal?

# Potential solution
Supplementary data

ptrbortolotti, 2023-10-25:
Please take a look at the FAQs 

https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#are-the-cfrp-properties-wrong

This journal article might come useful https://wes.copernicus.org/preprints/wes-2021-74/wes-2021-74.pdf

ncepulibei, 2023-10-26:
I saw that there are only two data in the journal article, and the yml file（https://github.com/NREL/BAR_Designs/blob/3a5a730a3ae965054e7f06549a8eb47caf6a728a/BAR_USC/BAR_USC.yaml#L818 ）Are these two zeros in lines 826 and 827 normal? 
Xt: [1285200000.0, 3810000.0, 0.0]
Xc: [878210000.0, 82180000.0, 0.0]

ptrbortolotti, 2023-10-26:
unfortunately we don't have data in direction 3 (through the laminate thickness)

ncepulibei, 2023-10-26:
Without data, is it impossible to perform failure analysis? Because the failure criterion in the figure below requires nine parameters.
[image]

ptrbortolotti, 2023-11-06:
you should still be able to perform the stress analysis, simply neglect the contributions from direction 3, which are hard to characterize and shouldn't matter much in hollow structures such as wind turbine blades

closed 2023-11-08 by ptrbortolotti