IEAWindSystems/IEA-15-240-RWT #170: Confirmation of Mooring Line Diameter for the UMaineSemi Model

LeoHongLiZuo, 2023-10-27:
Dear guys,
Mooring chain diameters in the Moordyn module of Openfast in the UMaineSemi model is 0.333m as show in the below figure.
[image]
But the diameter of the UMaineSemi model in the official definition file
 ' Definition of the UMaine VolturnUS-S Reference Platform Developed for the IEA Wind 15-Megawatt Offshore Reference Wind Turbine ',
as shown in the following figure, is 0.185 m.
[image]
So is the mooring chain diameter of the UMaineSemi model 0.333 m or 0.185 m ?
Best wishes!
Leo

mattEhall, 2023-10-27:
Hi Leo, we may follow up with this in more detail, but a quick note for now: chain size is typically referred to as "nominal diameter", which means the diameter of the steel bar that is used to form the chain link. But most of our models expect an input of "volume equivalent diameter", which is the diameter of a cylinder that would have displace the same volume as the chain. In this example, 185 mm is the nominal, and 333 mm is the volume-equivalent, so they are both right.
Matt

LeoHongLiZuo, 2023-10-28:
Dear Matt,
Thank you very much for your patient answer.
I would also like to ask how the specific numerical conversion relationship or formula between "nominal diameter" and "volume equivalent diameter" is set.
I didn 't find a specific description of the formula in "MoorDyn User’s Guide".
Because in the follow-up study, I may use other diameter mooring chains.
Best wishes!
Leo

mattEhall, 2023-11-02:
Sorry for the delayed response! For studless chain we have been using a ratio of d_volume = 1.8*d_nom  (as just posted here https://github.com/OpenFAST/openfast/issues/1842)
Matt

LeoHongLiZuo, 2023-11-03:
Dear mattEhall,
I 'm sorry I also asked on the Openfast Issues four days ago, because the experiment needs to take into account other diameters, so it 's a little urgent.
Thank you very much for replying me patiently in the busy work!
 Best wishes!
 Leo

closed 2024-07-11 by gbarter