IEAWindSystems/IEA-15-240-RWT #171: starting simulation of IEA-15MW RWT UMaineSemi

falascao, 2023-11-08:
Hello,
i'm trying to start simulation of IEA-15MW RWT UMaineSemi without change any parameters.
I have downloaded Rosco, Wisdem and BasicDtu controller.
[image]

This is the error I encountered:
[image]
Best regards,

ptrbortolotti, 2023-11-08:
Hello, in the ServoDyn file, you need to update the path to the ROSCO compiled file libdiscon

gbarter, 2023-11-08:
This [line](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ServoDyn.dat#L77) must be modified to point to the ROSCO dll file with your specific path.

falascao, 2023-11-08:
Thank you, it worked.
Best regards,

closed 2023-11-08 by ptrbortolotti