IEAWindSystems/IEA-15-240-RWT #173: simulation error 15MW monopile with comphydro deactivated

falascao, 2023-11-23:
Hi,
i'm trying to do simulations with the monopile 15MW wind turbine with comphydro deactivated and with stationary wind speed as input (10.75m/s).
The simulation response error is the following: 
[image]

I have neither recived the message "aborting fast" so the simulation doesn't stop.

gbarter, 2023-11-27:
Hello- You will probably have to post the files either here or on the [OpenFAST forum](https://forums.nrel.gov/top?period=quarterly) for us to be able to determine what is going wrong.

falascao, 2023-11-28:
Thank you very much, I solved the problem.
Best regards,

closed 2023-11-28 by falascao