IEAWindSystems/IEA-15-240-RWT #174: BModes .bmi and _tower_secs.dat files for the tower of the UMaine SS

maucollu, 2023-11-25:
## Description
This is not really a bug, but a request to make available (or to indicate where to find) the tower .bmi file and .dat file to run BModes, if possible.
I found a version here (link)[https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/v1.1.2/OpenFAST/IEA-15-240-RWT/BModes/IEA-15-240-RWT_BModes_tower.bmi], but the values indicated do not seem to coincide with the values in the report... as also indicated by Jason with his doubts about the correctness of the values.

I would like to use them for modifying the tower sections/thicknesses and derive the impact on natural frequency.

### Steps to reproduce issue
N/A

### Current behavior
Files not available

### Expected behavior
N/A

gbarter, 2023-11-27:
Please see a related post and answer on the OpenFAST forums [here](https://forums.nrel.gov/t/15mw-iea-rwt-b-mode-input-data/5966/4).

maucollu, 2023-11-27:
Many thanks, @gbarter.
Sorry, I realised that I did not explain myself very well -  I meant the BMOdes input files for the tower for the floating wind turbine, the UMaine SS. The ones in the links above seem to be the ones for the monopile if I'm not mistaken...

gbarter, 2023-11-27:
I am not aware of BModes input files for the UMaine semisub, sorry.  I would also be suspicious if there were any files out there as we had to redesign the floating tower design with more modern analysis tools to avoid resonance due to errors in the original design workflow.  Those changes were part of the v1.1 release.

I believe you would also have to use a non-standard version of BModes that is available only in the OpenFAST forums to properly account for the floating boundary condition.

maucollu, 2023-11-28:
Many thanks, @gbarter .

Yes, I am aware that a different version is needed, and if useful details and repository of the BModes version "JJ" is mentioned in this NREL Foum [post](https://forums.nrel.gov/t/elastodyn-bmodes-tower-input-file-and-mode-shapes-calculation/4127/5).

Would you know/be able to let me know who desigend the tower for the IEA UMaine SS, if possible?

Kind Regards

Maurizio

gbarter, 2023-11-28:
I was heavily involved in the revised floating tower design, which is "stiff-stiff".

maucollu, 2023-11-28:
That's great! Would it be possible to have access to the finalised .bmi and tower_data.dat used?

gbarter, 2023-11-28:
Since BModes is "deprecated" (not open-source and no ongoing development), it was not involved in the design process at all and we have pretty much removed it from our internal workflow.  The updated floating tower was designed in [WISDEM](https://github.com/WISDEM/WISDEM) and run in OpenFAST via the new-ish [WEIS](https://github.com/WISDEM/WEIS) tool.

You can find the diameter and thickness schedule of the floating tower [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT_VolturnUS-S.yaml#L508-L534).  You could use standard tubular cross-section theory to build up the BModes inputs yourself if that is something you are up for.

maucollu, 2023-11-28:
Many thanks, @gbarter , makes perfect sense to use a maintained software.
I know of WISDEM/WEIS, although I am not that familiar with it myself...but, looking at the manual of WISDEM, and TowerSE in particular [here](https://wisdem.readthedocs.io/en/master/wisdem/towerse/theory.html), it is mentioned that "Dynamics for floating turbines are not included in TowerSE."

So maybe you used another approach than TowerSE, or twicked TowerSE for the UMaine?

By the way, thanks a lot for pointing me to the exact place where to find the most up to date UMaine SS tower dimensions.

gbarter, 2023-11-29:
Thank you for pointing that out.  That is probably an over-simplification in the documentation that we need to edit.  The TowerSE module in WISDEM is indeed meant for applications with a clamped based (fixed-bottom offshore or land-based).  However, the larger WISDEM code has the FloatingSE module that has its own [tower modal analysis routine](https://github.com/WISDEM/WISDEM/blob/master/wisdem/floatingse/floating_frame.py#L300) that can handle the modified boundary condition.  Which routine is called should mostly be something the user doesn't have to worry about.

closed 2024-07-11 by gbarter