IEAWindSystems/IEA-15-240-RWT #175: Damping ratios UMaine VoluturnUS-S FOWT

andreowhite, 2023-12-15:
Hi everyone:
Does anyone know where I can find the damping ratios for UMaine VolturnUS-S FOWT?

Thank you.

Regards,
AOAW

mattEhall, 2023-12-16:
Hi Andre, can you be more specific please?

andreowhite, 2023-12-18:
Hi Matthew,
I would like to find out what are the values of the damping ratios for the different degrees of freedom (surge, heave etc.) of the platform.

Regards,
AOAW

[https://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

andreowhite, 2023-12-19:
Hi @mattEhall just checking if I am now clear. 
In the paper by Allen et al. (2020), which defined the UMaine VolturnUS-S platform, there is a graph of damping ratio vs initial cycle amplitude but I do not know if it possible to extract the damping ratio for each mode of vibration from that and I am not sure what initial cycle amplitude is. All I would like is to find out damping ratio for surge, heave and pitch DOFs.
Hope I am clear now.

Regards,
AOAW

ptrbortolotti, 2024-11-21:
@mattEhall @gbarter we never got back to @andreowhite unfortunately. is this issue still relevant, or shall we close it?

closed 2025-04-10 by ptrbortolotti