IEAWindSystems/IEA-15-240-RWT #176: IEA-15-240-RWT-UMaineSemi Operational issues

sffgd, 2023-12-26:
Dear author，
      I get the error "FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library/usr/share/miniconda3/envs/test/lib/libdiscon.so could not be loaded. Check that the file exists in the specified location and that it is compiled for 64-bit applications.”when I run IEA-15-240-RWT-UMaineSemi
     I used the openfast_x64.exe files from openfast 3.0.0 and openfast 3.5.1 respectively, and I wondered if it was due to this error. Thank you very much for the answer to the hope!

dzalkind, 2023-12-26:
Hi, the error states that the libdiscon.so file (the ROSCO dynamic library) required to run this OpenFAST model may not exist.  You can download one for your operating system here: https://github.com/NREL/ROSCO/releases/tag/v2.8.0

sffgd, 2023-12-27:
Thank you for your quick answer！

sffgd, 2023-12-27:
I've downloaded this libdiscon.so file,But I haven't been able to find this location to put libdiscon.so（The dynamic library/usr/share/miniconda3/envs/test/lib/libdiscon.so）

dzalkind, 2023-12-27:
You can put the file wherever you like and add the path to the ServoDyn input `DLL_InFile`.  It looks like you're using windows, so you'll want a dynamic library with a .dll extension.

sffgd, 2023-12-28:
Thank you for your answer. This question has been resolved. Additionally, I would like to ask you a question about simulating a wind turbine shutdown state. How can I modify the parameters in the file specifically?

gbarter, 2024-07-11:
Sorry that this last question wasn't addressed.  I believe that is better directed towards the OpenFAST [Forums](https://forums.nrel.gov/).  If I do a search there for "shutdown", I get a handful of threads that should be a helpful guide.

closed 2024-07-11 by gbarter