IEAWindSystems/IEA-15-240-RWT #177: QTF of UMaine VolturnUS-S platform calculated by Nemoh is inconsistent with the results given by WAMIT.

getChaos22, 2023-12-27:
Hello everyone,

Recently I am trying to simulate the first order and second order hydrodynamics of UMaine VolturnUS-S using [Nemoh](https://gitlab.com/lheea/Nemoh) (an opensource boundary element method solver developed by Ecole Centrale de Nantes). I take the result of WAMIT as a reference to verify the model I constructed. The first order hydrodynamics calculated by Nemoh match very well with the result of WAMIT. However, there is a difference between QTF obtained by Nemoh and WAMIT. The followings are the comparison of difference QTF from Nemoh and WAMIT.

[image]

Do you have any ideas or advises on the the differences?

The attachment is the .cal file I used to run the simulation. Mesh of the platform is converted from the WAMIT mesh obtained from [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData/wamit_inputs_2ndOrder). The inertia matrix and additional stiffness matrix that is needed to calculated the platform motion are also taken from [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData/wamit_inputs_2ndOrder).

Thank you!

[Nemoh.zip](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/13776495/Nemoh.zip)

luwang00, 2023-12-27:
Hi @getChaos22,

Thanks for bringing this to our attention. I'm not familiar with Nemoh, so it is difficult to say what exactly is causing the difference. One thing I noticed is that you did not include free-surface forcing in your calculation. This might have some effect on the results.

Another consideration is that with difference-frequency QTFs, usually the only values that matter are the ones within a narrow diagonal band in the contour maps (the frequency pairs giving a low difference frequency below those of linear wave excitation). Therefore, instead of comparing a horizontal cut, it might be a better idea to compare the values along diagonal cuts with difference frequencies close to zero. Further away from the main diagonal, it becomes increasingly challenging to get accurate results, but those values usually don't matter in practice.

getChaos22, 2023-12-28:
Dear @luwang00 ,

Thanks for your quick response and advises. The information is important to me. 

I did not include free-surface mesh at first because according to Nemoh's manual, effect of the free surface mainly contributes to sum-frequency QTF, it may be negligible for difference-frequency QTF calculation. But I will include free-surface to figure out the effect on difference-frequency QTF this time.

The data on the diagonal do match with WAMIT better. The followings are the difference-frequency  QTF on the diagonal. 

[image]

On the DoF of heave and pitch, results of Nemoh coincide with WAMIT quick well, but shows a larger difference at surge direction. The results did not account for the effect of free-surface. I'll construct free-surface mesh and see the difference.

Thanks again!

Chao Li

getChaos22, 2024-01-02:
Hi @luwang00,

Last week, I included a free-surface mesh in the simulation of QTF for UMaine platform. The result of sum-frequency is better than the case that did not contain free-surface effect, but for difference-frequency, the result did not change a lot. 

[image]

It seems like the free-surface mainly affect the calculation of sum-difference QTF in Nemoh. I still did not figure out why the difference of the difference-frequency QTF between WAMIT and Nemoh.

I notice the frequency range employed to calculate first order hydrodynamics has a large effect of the second order hydrodynamics. But I hasn't tuned a suitable frequency range that can match the difference-frequency QTF of Nemoh with WAMIT.

Do you have some suggestions of the frequency range that used to calculated the first order hydrodynamics?

Thanks!

Chao Li

luwang00, 2024-01-04:
Hi @getChaos22,

Sorry of the delayed reply. I'm glad to see the difference-frequency QTFs in heave and pitch are mostly consistent near the diagonal. The difference in the surge force is mostly at higher frequencies where we usually don't expect significant wave presence.

Regarding the mean surge force, I think what you got from Nemoh might be more accurate than the WAMIT solution currently available from this repository. We have recently regenerated the QTFs for VolturnUS using the higher-order panel method of WAMIT. The values along the diagonal are plotted below. It looks more consistent with your results from Nemoh. The original QTF values from WAMIT were computed using the low-order panel method, so perhaps the convergence is not as good at higher wave frequencies.
[image]

It will likely take us some time to update the QTFs here, but in the meantime, I think it is reasonable to trust your Nemoh solution. 
Regarding the first-order frequency range to use when computing the QTFs, usually it just has to be wide enough to cover the wave frequency range of the incident wave spectrum.

getChaos22, 2024-01-05:
Hi @luwang00,

Thanks for sharing the newest result. This is very useful to me. 
Looking forward to see the final QTF results.

Best Regards

Chao Li

closed 2024-07-11 by gbarter