IEAWindSystems/IEA-15-240-RWT #178: Time histories of free decay tests for each DoF

SMSJHHuang, 2024-01-05:
Hello,

I am trying to validate the correction of the FOWT model that we built in Sesam Sima using free decay tests. That means that we could confirm the correction of our models by comparing the free decay simulated results and NREL technical report. Could we get the relevant time series results?

Thanks

gbarter, 2024-01-06:
Hello,

We do not maintain an active list of results that is always current with the latest OpenFAST/HAWC2 releases or model updates in this repository.  Similarly, the results that were published in the original report may not be perfectly recoverable with the current model.  The simulation outputs that were used in the report are also not available.  Our best advice is for users to re-run the simulations for themselves.

Apologies for the additional work.

SMSJHHuang, 2024-01-25:
Dear @gbarter 
Thank you for answering our question. We attempted to reproduce the time series using  the website “http://www.graphreader.com/,“ and it works.

Best.

closed 2024-01-25 by SMSJHHuang