IEAWindSystems/IEA-15-240-RWT #179: Deviations between the 6*6 stiffness through SONATA results and the origin BeamDyn blade data

winder365, 2024-01-11:
Dear @ptrbortolotti 

Sorry to disturb you. 

I am trying to use SONATA/ANBA4  to calculate the 6*6 stiffness and mass matrixes of the IEA-15MW blade, and I have known that the 6*6 matrixes of BeamDyn was calculated through SONATA/ANBA4 (https://github.com/IEAWindTask37/IEA-15-240-RWT/issues/96#issuecomment-1222895729). 

However, when I tried to calculate the 6*6 matrixes through SONATA/ANBA4, two problems occured. The result I got through SONATA/ANBA4 (https://github.com/ptrbortolotti/SONATA/tree/master/examples/1_IEA15MW) is different from that of the original BeamDyn blade data (from https://github.com/IEAWindTask37/IEA-15-240-RWT/releases/tag/v1.1.4), and this lead to deviation of the rotor load (y axis is shown as log axis):

[image]

[image]

[image]

I want to know how to get the same result as the original BeamDyn blade data.

Thank for your attention

winder365, 2024-01-27:
Dear @gbarter and @ptrbortolotti 

Sorry to disturb you.
Is there any solution available?

Thank for your attention.

ptrbortolotti, 2024-01-27:
Hello,
my apologies, I completely missed this open issue. I believe the issue is related to the detailed modeling and meshing of the trailing edge, which has changing in SONATA over the years. The original blade was designed in PreComp, which adopts a shell formulation and adopts several simplifications when it comes to TE modeling. SONATA (and ANBA4) adopt 2D planar elements and allow to model the TE much more precisely. However the modeling of the TE of the 15MW was totally arbitrary, since it was never designed against fatigue loads, bond line cracking, or manufacturing constraints. Overall if I were you I would use the new results if you were interested in the details of the blade structure, since you have it under control in SONATA, or the older results if you are more interested in staying consistent with the IEA15 repo. Since at this point we are more interested in the latter aspect, I do not anticipate changes to the BeamDyn model.
I hope this helps!
Best regards,
Pietro

winder365, 2024-01-27:
Dear @ptrbortolotti, 

Thank you for your reply. I still have two question:

(1)  What I wandered is that why the difference in section property of the blade has resulted in significant difference in rotor power:

[image]

(2)  Besides, according to your reply, I understand that the section property used in BeamDyn now are calculated by PreComp right?

Thank you for your attention.

winder365, 2024-02-21:
Dear @ptrbortolotti 

As you instruction, I have look through the configuration file of the OpenFAST, and I didn't open the servo-control or set prescribed pitch during the simulation,  and the simulation conducted without changing the BeamDyn blade file are all OK.

The version of the SONATA I used is from “https://github.com/ptrbortolotti/SONATA”, and the .yaml file provided in the example folder was used to calculate the 6*6 matrix, however the matrixes calculated are different from the BeamDyn file what provided in IEA-15MW package "https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat".

As mentioned in the above post, the torsional and edgewise stiffnesses I calculated through SONATA/ANBA4 are large than that provided in the BeamDyn file, whcih lead to the difference in power and thrust of the rotor.  Is there any bugs in the .yaml file or the SONATA program I used?

I need some help to calculate the correct blade section property, looking forward to you instruction.

Thank for your attention.

ptrbortolotti, 2024-02-27:
Hello,
thank you for your persistence and patience. I've run a few tests with OpenFAST 3.5.2 and I'm a little puzzled by the results. The results with the BeamDyn files from the latest SONATA make sense to me, whereas the ones with the BeamDyn files from the repo seem to go unstable, not yet sure whether that's numerical or physical. I would trust your newer files while we keep debugging this.
I hope this helps.

winder365, 2024-02-28:
Dear @ptrbortolotti 

Thank for you reply and patience.

The reasons for my concentration on the blade section property is that I want to conduct some lay-up modification research of  the IEA-15MW blade, and I need to found a correct (or proper) tool for blade section property calculation.

To ensure the method the or the tool I use for blade section property calculation is correct (or proper), the first thing I need to do is to reproduce the blade property used in BeamDyn file what provided in IEA-15MW repository (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat). However, I can not reproduce that now.

Can you provide me the ".yaml" file (used for SONATA) file which can be used to obtained the same blade section property as that in the BeamDyn file, and the corresponding version package of the SONATA (as you mentioned above the current BD file was generated using an older version of SONATA).

Besides, the ".yaml" file provided in the IEA-15MW repository may be not used directly for SONATA, as there are some bug report. The  ".yaml" file provided in the SONATA repository run OK, but the results are not agree with the IEA-15MW's BD file.

Thank for you patience and reply again, looking forward to your help.

ptrbortolotti, 2024-03-05:
I've been making some progress on this issue. The example from SONATA had the c2_axis flag mistakenly turned on. The flag switches the reference frame of BeamDyn between the blade reference axis (what's in this repo) and the mid-chord axis (used by HAWC2). I've now turned off the flag in this commit https://github.com/ptrbortolotti/SONATA/commit/ef9a474064b66e4a248da28cb77910528b2ca3fa
If you update and rerun SONATA, you should now see much smaller differences in the K and I matrices. Note that you can also stick to the mid-chord definition of BeamDyn, but then you need to update the x, y, and z points in the main BeamDyn file. FYI, BeamDyn accepts any generic reference axis, as long as K and I matrices are defined with respect to that reference axis.
I will also add that with steady blade aerodynamics and BeamDyn, the rotor seems to be quite unstable in torsion. The rotor was not designed at the level of fidelity of BeamDyn, and we might have some stability issues. I'm still working on this and will share an additional update in the next days

winder365, 2024-03-05:
Dear @ptrbortolotti 

Thank you for your endeavor on this issue very much!

I have update the SONATA@[ef9a474](https://github.com/ptrbortolotti/SONATA/commit/ef9a474064b66e4a248da28cb77910528b2ca3fa), and conducted some tests. As you say, now there are smaller differences in the K and I matrices and the load calculation results through the OpenFAST are more reasonable.

Thank for your endeavor and help again, best wishes!

ptrbortolotti, 2024-03-19:
hello again,
quick update. The IEA15 OpenFAST model looks rather unstable when steady airfoil aerodynamic models are used. Unsteady airfoil aero models make the model stable. So in doubt, I'd opt for the latter. Does this help? More results will be presented at TORQUE 2024

closed 2024-04-05 by ptrbortolotti

winder365, 2024-04-10:
Dear @ptrbortolotti 

Thank you for your effort. You are right, if we turn off the UA model, the blade seems enter the flutter state even at the rated condition. As mentioned in Hansen's "Aerodynamics of wind turbines", the aeroelastic simulation with the UA are more realistic.

Looking forward to your report in TORQUE 2024.

Thank you for your effort again！

ptrbortolotti, 2026-02-24:
hi, we should have been more rigorous about tagging SONATA as it progressed. I've just checked the repo, and the K and I matrices in the BeamDyn file have not changed since 2021 (you can check this yourself as well). you could go back to those days in the development of SONATA, and replicate that file. however, I don't think I recommend this approach. rather, I'd run the latest SONATA, and take the output BeamDyn file as your new reference

please note that in the development reference turbines we favor stability over small improvements. unless we find something substantially wrong in the files, we deliberately chose not to update them to maximize the reproducibility of the model across studies (we could debate if this choice is wise, but that's how things are today)

overall, I think the differences that you show are somewhat in line with what I expect from slight differences in meshing of the cross sections. I suspect that if you were to compute the natural frequencies of the old and new blades, they would come out very close to each other