IEAWindSystems/IEA-15-240-RWT #180: A question about  IEA-15-240-RWT-UMaineSemi in OpenFAST

LeoHongLiZuo, 2024-02-02:
Dear guys
I have a question about IEA-15-240-RWT-UMaineSemi in OpenFAST.
As shown in the following figure, I want to add appropriate AXIAL COEFFICIENTS (such as AxCd) to the floating platform in HydroDyn.
[image]
But I have not found the appropriate value, does anyone know how to define the parameter there.
Thanks a lot!

luwang00, 2024-02-02:
Hi @LeoHongLiZuo,

In the HydroDyn model shared for the UMaineSemi, there are already quadratic damping coefficients specified (see AddBQuad). These are used to represent the viscous damping in this model, so no drag coefficient is provided. In fact, there isn't any strip-theory member under MEMBERS that actually uses the drag coefficient you are referring to. 

Are you defining additional strip-theory members that require these coefficients? In general, while it is possible to have a reasonable guess for the added mass and pressure coefficients, the drag coefficients usually need to be tuned to either CFD simulations or some experimental measurements.

LeoHongLiZuo, 2024-02-03:
Dear @luwang00 
Thank you very much for your advice!
I think I still need to set a suitable resistance parameter.
If it is convenient for you, we can add a wechat communication,13314719037.
Looking forward to your reply!
Yours,
Leo

luwang00, 2024-02-07:
Hi @LeoHongLiZuo,

I'm afraid we can only support answering questions through GitHub. Thank you for your understanding.

closed 2024-04-28 by ptrbortolotti