IEAWindSystems/IEA-15-240-RWT #186: Mismatchbetween provided Cp,Ct, Cq data in IEA-15-240-RWT and data generated using AeroDyn_Driver.exe

MuhammadSharjil, 2024-02-15:
Hi
I have a query. I have used AeroDyn_Driver_x64_Double.exe utility provided with OpenFast to generate Cp, Ct, Cq data against TSR and Blade Pitch Angle for IEA-15-240-RWT-UMaineSemi case.
I am facing an issue, the values doesnt match between:
a) [IEA-15-240-RWT/OpenFAST/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt at master · IEAWindTask37/IEA-15-240-RWT · GitHub](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt)
b) Data generated using AeroDyn_Driver_x64_Double.exe utility.
Cp value comparison at, TSR = 9, with Blade Pitch Angle = 0,

(for a), Cp = 0.4692
(for b), Cp = 0.4871
(Moreover, this Cp value at TSR = 9 & Blade Pitch Angle = 0, is maximum for the Cp data I have generated with Aerodyn_Driver.)

ad_driver.dvr
[image]

IEA-15-240-RWT-UMaineSemi_AeroDyn15.dat
[image]

Can you help me in figuring out what settings I need to change?
Thanks

Regards,
Muhammad Sharjil

ptrbortolotti, 2024-02-15:
Hello Muhammad,
those tables were generated with CCBlade, not with OpenFAST. CCBlade is the steady-state BEM solver of WISDEM https://github.com/WISDEM/WISDEM
I have never ran a detailed code2code comparison between AD15 and CCBlade. The formulations are similar, but there are certainly some differences between the two codes.
Best regards,

MuhammadSharjil, 2024-02-19:
Hi @ptrbortolotti  for the response.
I found certain differences between the two below items
a. https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt
b. https://www.nrel.gov/docs/fy20osti/76773.pdf
The Cp and Ct doesnt matches in above cases.
I want to get an opinion about what value to use for calculation of region-2 Kw2 gain and for any other analysis.
Thanks

Regards,
Muhammad Sharjil

gbarter, 2024-02-19:
Hello Muhammad,
The reference wind turbine definition has seen many bug fixes, as well as improvements to the computational models, since the original report.  We consider simulations using the latest model files to be "truth".  We also can no longer recover some of the results shown in the report due to these changes.  Some similar comments are made in the [FAQ](https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#i-get-a-different-rated-thrust-value-than-the-report).

Cheers,
Garrett

closed 2024-07-11 by gbarter