IEAWindSystems/IEA-15-240-RWT #187: Yaw-control

maucollu, 2024-02-16:
# Description of feature
Hi, I was wondering if the IEA 15MW wind turbine model of the UMaine SS comes with a yaw control system that can be switched on. I noticed that this is switched off, and therefore the yaw DOF of the nacelle should be switched off, and the yaw nacelle angle aligned with the wind direction.

Anyway, the platform may yaw during the simulation, and therefore there will be a skew angle between the rotor and the nacelle,.

I tried:
- in Elastodyn.dat: YawDOF True
- in Servodyn.dat: YCMode 5, TYCOn 0.0, and leaving all the others the same

but nothing seems to change in the simulation (that is even stranger, since I would have expected either a yaw control to kick in, or the simulation to crash since the control model was not working).

# Potential solution
I was wondering if I am doing something wrong, or this feature was not implemented?

gbarter, 2024-07-11:
Sorry that this fell through the cracks.  I will look into this and get back to you

dzalkind, 2024-07-11:
Maurizio and I discussed the following over email that may be helpful to others in the future:

ROSCO has a yaw controller that can be activated here: [https://github.com/NREL/ROSCO/blob/7deb583d4acbacc068bad28e2228f7d257a7cd7c/Examples/Test_Cases/IEA-15-240-RWT-UMaineSemi/DISCON-UMaineSemi.IN#L16](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNREL%2FROSCO%2Fblob%2F7deb583d4acbacc068bad28e2228f7d257a7cd7c%2FExamples%2FTest_Cases%2FIEA-15-240-RWT-UMaineSemi%2FDISCON-UMaineSemi.IN%23L16&data=05%7C02%7CDaniel.Zalkind%40nrel.gov%7Cff73a01be5fc42d6c28a08dc870e0ff9%7Ca0f29d7e28cd4f5484427885aee7c080%7C0%7C0%7C638533741721365096%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=vcTtap3UW5fGvOb%2F%2FRMqvg6mmpjsiSdlTjrcCe7iuhM%3D&reserved=0)
 
With these parameters: [https://github.com/NREL/ROSCO/blob/7deb583d4acbacc068bad28e2228f7d257a7cd7c/Examples/Test_Cases/IEA-15-240-RWT-UMaineSemi/DISCON-UMaineSemi.IN#L118](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNREL%2FROSCO%2Fblob%2F7deb583d4acbacc068bad28e2228f7d257a7cd7c%2FExamples%2FTest_Cases%2FIEA-15-240-RWT-UMaineSemi%2FDISCON-UMaineSemi.IN%23L118&data=05%7C02%7CDaniel.Zalkind%40nrel.gov%7Cff73a01be5fc42d6c28a08dc870e0ff9%7Ca0f29d7e28cd4f5484427885aee7c080%7C0%7C0%7C638533741721373412%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=JLAEZZRXwJuBCi7aJKwoTIx2K5m61tvvO29lyd55%2Fpk%3D&reserved=0)
 
There isn't a lot of documentation on the yaw controller, but the source code is here:
[https://github.com/NREL/ROSCO/blob/7deb583d4acbacc068bad28e2228f7d257a7cd7c/rosco/controller/src/Controllers.f90#L310](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNREL%2FROSCO%2Fblob%2F7deb583d4acbacc068bad28e2228f7d257a7cd7c%2Frosco%2Fcontroller%2Fsrc%2FControllers.f90%23L310&data=05%7C02%7CDaniel.Zalkind%40nrel.gov%7Cff73a01be5fc42d6c28a08dc870e0ff9%7Ca0f29d7e28cd4f5484427885aee7c080%7C0%7C0%7C638533741721379097%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=IRqbhUFGHOrUqPqvhop4xi8QdaGMrJB9SmqU23G3gmI%3D&reserved=0)
 
You also need to make sure that YawDOF is True in ElastoDyn and YCMode = 5 in ServoDyn.

closed 2024-11-21 by ptrbortolotti