IEAWindSystems/IEA-15-240-RWT #191: I met some

ZHaaaAO, 2024-02-29:


ZHaaaAO, 2024-04-02:
Dear @ptrbortolotti ：

I hope this message finds you well. I wanted to bring to your attention an error that I discovered in the previous modeling work. **It appears that I overlooked the continuity of changes between the different sections. I apologize for this oversight.**

Now, in order to rectify this error, I need to obtain the data of the layup material at different sections for input in Precomp. However, upon reviewing the report, **I noticed that it only provides material data at a few specific locations**, such as 24.517% span and 77% span. Unfortunately, **it does not provide data for all the required locations.**

**I was wondering if there is any way to obtain the material data for all the sections in order to ensure accurate modeling and analysis**. Your guidance and assistance in this matter would be greatly appreciated.

Thank you for your attention to this matter.

Best regards,
@ZHaaaAO

ptrbortolotti, 2024-04-03:
hello @ZHaaaAO , my apologies for not replying earlier, I missed this notification among my emails.
let me see if I understand your request. are you looking for cross sectional geometries and material properties? if so, these are specified in this .yaml file https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT.yaml following the windIO convention, which is documented here https://windio.readthedocs.io/en/latest/
the yaml file is read into https://github.com/WISDEM/WISDEM, which calls the PreComp 2D cross sectional solver. if you feel like, you can compile WISDEM and run example 02, which will run the IEA-15 turbine here https://github.com/WISDEM/WISDEM/tree/master/examples/02_reference_turbines
Note that PreComp generates the 4x4 stiffness matrix, whereas BeamDyn requires the 6x6 matrix. To generate the BeamDyn file, we ran this model https://github.com/ptrbortolotti/SONATA/tree/master. example https://github.com/ptrbortolotti/SONATA/tree/master/examples/1_IEA15MW runs the IEA15
I hope this helps

ZHaaaAO, 2024-04-03:
Dear @ptrbortolotti 
Thank you so much for providing this information! 
I greatly appreciate it, as it has resolved my question. I'm glad to know where I can find the cross-sectional geometries and material properties in the specified .yaml file following the windIO convention. The documentation you shared will be very helpful.

I also want to express my gratitude for guiding me to the WISDEM repository, where I can compile and run the example with the IEA-15 turbine. 

 Thank you for your assistance！I will proceed with the instructions provided.

rngrose, 2024-05-29:
Dear @ptrbortolotti

When I extract the material information from "IEA-15-240-RWT_tabular.xlsx" and "IEA-15-240-RWT.yaml" and use it into "materials.inp" of Precomp, I find some differences that confuse me. 

In "IEA-15-240-RWT_tabular.xlsx", the shear module data is provided in the format of "**G_1,G_2,G_3**", the poisson ratio data is provided in the format of "**Nu_1,Nu_2,Nu_3**". (In the .yaml file, the data is provided in the same matrix format.)

But in the input file "materials.inp" of Precomp, the required data is "**G_12,Nu_12**".  

I have looked up some book about composite materials rapidly, **but I haven't found clear conversion relationship between "G_1,G_2,G_3,Nu_1,Nu_2,Nu_3" and "G_12,Nu_12".** I also consult the reference mentioned in "IEA-15-240-RWT_tabular.xlsx", which is "MSU composites database 3D property tests, Engineering Mechanics of Composite Materials, Daniel, I & Ishai, O., 1994, pg. 34", but I can't find some useful guidelines.

I know the material data is so important in the cross-sectional structural modeling and mechanical analysis, so I want to know how to obtain exact PreComp-needed material data based on the "IEA-15-240-RWT_tabular.xlsx" or "IEA-15-240-RWT.yaml"      

Sincerely hope for your reply.

rngrose, 2024-05-29:
> I think this is a copy-paste error in the header of the Excel file. and are properties of shear, which by definition is across two principal dimensions. So, where it says , it should be and same with . The order listed is G_12, G_13, G_23. I will fix that in the scripts that generate that table.`G``nu``G_1``G_12``nu`

Thank you sooooo much!! Your reply is important for me.

Best wishes
Wu

closed 2024-07-11 by gbarter