IEAWindSystems/IEA-15-240-RWT #196: Tower strike

locloc408, 2024-04-08:
Hello i don't know how to fix the tower strike when i run for the DLC6.1 with wind speed of 42.6mps  , would you guy have any reccomendation for my case

OpenFAST

Copyright (C) 2023 National Renewable Energy Laboratory
Copyright (C) 2023 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.5.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 2021
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: May 17 2023
 - Time: 13:43:33
Execution Info:
 - Date: 04/08/2024
 - Time: 12:29:28+0700

OpenFAST input file heading:
    IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

Running ElastoDyn.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.

   Reading a 31x31 grid (282 m wide, 9 m to 291 m above ground) with a characteristic wind speed
   of 50 m/s. This full-field file was generated by TurbSim (from OpenFAST-v3.4.1) on 08-Apr-2024
   at 11:08:58.

   Processed 36000 time steps of 20-Hz grid-field data (period of 1800 seconds).
Running HydroDyn.
  WaveTMax is less then the simulation time.  Wave data will repeat every WaveTMax seconds.
 Generating incident wave kinematics and current time history.
 Calculating second order difference frequency wave kinematics.
 Calculating second order sum frequency wave kinematics.
 Reading in WAMIT output with root name
 "C:\Users\Nam\Desktop\OpenFAST_BASE\Loc\HydroData/IEA-15-240-RWT-UMaineSemi".
 Computing radiation impulse response functions and wave diffraction forces.
 Calculating second order difference-frequency force using the full quadratic transfer function.
Running MoorDyn (v2.0.0, 2022-12-08).
   This is MoorDyn v2, with significant input file changes from v1.
   Copyright: (C) 2022 National Renewable Energy Laboratory, (C) 2019 Matt Hall
   Parsing MoorDyn input file:
   C:\Users\Nam\Desktop\OpenFAST_BASE\Loc\IEA-15-240-RWT-UMaineSemi_MoorDyn.dat
   Created mooring system:  3 lines, 6 points, 0 rods, 0 bodies.
   Finalizing initial conditions using dynamic relaxation.
   t=60  FairTen 1: 3.75334E+06, 3.67842E+06, 3.74616E+06
   Fairlead tensions did not converge within TMaxIC=60 seconds.
   MoorDyn initialization completed.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows).
Using legacy Bladed DLL interface.
 Time: 0 of 1500 seconds.
The BEM solution is being turned off due to low TSR.  (TSR = 1.897). This warning will not be

------------------------------------------------------------------------------
Running ROSCO-v2.6.0
A wind turbine controller framework for public use in the scientific field
Developed in collaboration: National Renewable Energy Laboratory
                            Delft University of Technology, The Netherlands
------------------------------------------------------------------------------
Generator speed:    7.6 RPM, Pitch angle:   1.0 deg, Power:     0.0 kW, Est. wind Speed:  54.3 m/s

FAST_Solution:FAST_AdvanceStates:AD_UpdateStates:BEMT_UpdateStates(node 47, blade
1):UA_UpdateStates:UA_UpdateDiscOtherState:ComputeKelvinChain:Mach number exceeds 0.3. Theory is
invalid. This warning will not be repeated though the condition may persist.
FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 5, blade 1):UA_CalcOutput:UA_BlendSteady:Temporarily turning off UA due to high angle of
attack or low relative velocity. This warning will not be repeated though the condition may
persist.

FAST_Solution:FAST_AdvanceStates:ED_ABM4:ED_CalcContStateDeriv:SetCoordSy:Small angle assumption
violated in SUBROUTINE SmllRotTrans() due to a large blade deflection (ElastoDyn SetCoordSy). The
solution may be inaccurate. Simulation continuing, but future warnings from SmllRotTrans() will
be suppressed.
 Additional debugging message from SUBROUTINE SmllRotTrans(): 0.55 s

** WARNING: Tower strike. **  This warning will not be repeated though the condition may persist.

FAST_Solution:FAST_AdvanceStates:AD_UpdateStates:SetDisturbedInflow:TwrInfl:Tower strike.

 OpenFAST encountered an error at simulation time 1.1 of 1500 seconds.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

locloc408, 2024-04-08:
i have noticed that Openfast created characteristic wind speed of 50 m/s but in my case i want it only for 42.6 m/s , so is that the error from the Turbsim i guesss ?

ptrbortolotti, 2024-04-08:
There is something going on here. Are you trying to run a storm case (DLC 6.1?). Then why do you have initial rpm and ROSCO? You should probably revisit your initial condition in ElastoDyn, like 0rpm and pitch of 90 deg. Also, you won't need ROSCO

closed 2024-07-11 by gbarter