IEAWindSystems/IEA-15-240-RWT #197: A question about the LIDAR measurement output

DavidChen0429, 2024-04-15:
Hey, 

I am new to OpenFAST, and I am writing this to ask a question about the LIDAR measurement output. I wanted to visualize the LIDAR (sensorType: continuous) output of the IEA-15-240-RWT-Monopile turbine. When I didn't add any output the simulation ran successfully, but after adding "WindMeas1" in inflowFile.dat, I got an error like this:
[image]

Does it mean that the turbine is unable to visualize the LIDAR output? If so, is there an alternative way to visualize it?

B.R.
David

gbarter, 2024-04-15:
This question might be better direction at the [NREL Forums](https://forums.nrel.gov/), which also might be a good place to search for prior questions on the Lidar output capability.  I will forward your question along.

DavidChen0429, 2024-04-16:
Dear @jjonkman,

Thanks for the help, I did notice a typo when copying the wind sensor measurements parameter from the OutListParameters.xlsx, and now it finally works.

Maybe a follow-up question about the LIDAR, I am a bit overwhelmed by the parameters in the inflowfile.dat about LIDAR, I think it would be very helpful if there exists documentation about LIDAR in OpenFAST, do you happen to know where I can find it? 

Best regards,
David

jjonkman, 2024-04-17:
Dear @DavidChen0429 ,

I’m not aware of published documentation on the LiDAR functionality added to the InflowWind module in OpenFAST v3.5.  I would suggest posting your question(s) on the OpenFAST GitHub repository (https://github.com/OpenFAST/openfast/discussions) and hopefully one of the developers can answer them.  (I don’t personally have experience with this module myself.)

Best regards,

closed 2024-07-11 by gbarter