IEAWindSystems/IEA-15-240-RWT #199: Wrong nose CoM_TT_x in tabular data

verlivkra, 2024-04-23:
## Description
I am wondering if the center of mass of the nose relative to the global coordinate system listed in the excel sheet is wrong. As this is auto-generated by wisdem, I am worried that there are other issues related to the mass calculations of the nacelle as well, without having checked the other components. 

### Steps to reproduce issue
For the IEA 15 MW, I have calculated that the length of the nose should be 2.1 m. The nose cross-section doesn't vary with the length of the nose, hence the local CoM_x, with x-along the nose, should be 1.05 m. 

### Current behavior
Nose center of mass in the excel spreadsheet is CoM_TT_x = -5.4476 m and CoM_TT_z = 4.922 m.

### Expected behavior
With a horizontal distance from the tower top to the joint between the nose and the bedplate of 5 m, and a shaft tilt of 6 deg, the global CoM_TT_x of the nose should be -5 m - 1.05*cos(6 degree) = -6.044 m.
Similarly, CoM_TT_z should be H_bedplate + L_nose/2*sin(6 degree) = 4.875 + 1.05*sin(6 degree) = 4.985 m (minor difference in Z). 

### Code versions
_List versions only if relevant_
- Python
- …

gbarter, 2024-04-23:
Thank you for your critical thinking here!  I think the detail that is missing is that the nose if offset from the tower centerline by the bedplate.  See [this diagram](https://wisdem.readthedocs.io/en/master/_images/layout_diagram.png).

verlivkra, 2024-04-23:
Yes, and by adding the offset between tower centerline and the intersection between the nose and the bedplate (L_bedplate = 5 m) to the local center of mass of the nose, I expect to get: 
CoM_TT_x = -L_bedplate - L_nose/2*cos(ShaftTilt) = -5m - 2.1m/2*cos(6 degrees) = -6.044m, 
not CoM_TT_x = -5.4475 m. 
Or am I wrong?

gbarter, 2024-04-23:
Well, the plot thickens.  I agree with the correct answer and that the current tabular data isn't correct.  However, my PR earlier was too hasty and WISDEM was indeed calculating the correct value internally thanks to [this line](https://github.com/WISDEM/WISDEM/blob/master/wisdem/drivetrainse/drive_components.py#L888).  I am confused as to why the Excel data is wrong and will look into the auto-generation scripts.

gbarter, 2024-04-24:
I did find the culprit in some keywords in the yaml-files here that were not being read correctly.  There is a PR to fix the issue, but we'll have to do some due diligence on it first.

verlivkra, 2024-04-24:
Thanks for following up! While your at it, I also think the tabulated nose mass is wrong (assuming it is given in [kg] and not [kg/m]). 
I believe it should be 
$mass = L \rho \pi 1/4 (DO^2-DI^2) = 2.1 \times 7850 \times  3.14 \times 0.25 \times (2.2^2-2^2) = 10876$ kg. 
Not 6733 kg, as tabulated.

verlivkra, 2024-04-26:
In the same excel sheet, I also see that the generator stator and generator rotor have the exact same center of mass in the global coordinate system. Is this a reasonable assumption? They are offset in the pdf-documentation ("Definition of the IEA 15..."), but I don't know if those values are trustworthy either.

gbarter, 2024-04-26:
When writing the report, we had a detailed CAD model of the nacelle and all of the components.  That model had a number of issues that were raised later, which is why we don't recommend relying on the published table values in the report.  Since then, we've relied on the analytical models and do not have a detailed CAD model of the nacelle.  Our analytical model doesn't have the resolution on the generator to distinguish the generator-rotor/stator from one another, so their CoM are reported to be the same.

closed 2024-07-11 by gbarter