IEAWindSystems/IEA-15-240-RWT #203: Question about IEA-15-240-RTW v1.1.11

achilleas101, 2024-04-29:
I just downloaded version 1.1.11 of IEA-15-240-RTW and stumbled upon something I cound not understand in the IEA-15-240-RWT_tabular.xlsx. The Pitch-Wind speed plot in the .xlsx is not the same as in the 15MW wind turbine documentation [found here](https://www.nrel.gov/docs/fy20osti/75698.pdf).

[image]
Figure 1: Pitch-Wind speed from older version

[image]
Figure 2: Pitch-Wind speed from v.1.1.11

Am I missing something important?
What is causing such a drastic change of values?
Should I use these values as initial conditions in the ElastoDyn input file or stick to the old ones?

Thank you for your time and patience in advance!

gbarter, 2024-04-30:
Good catch!  The conditioning of the pitch optimization was poor near cut-in when power was near zero, so that caused things to go awry.  Not sure why that appeared now vs earlier.  PR #204 will fix the issue.

In general, we do not expect a perfect match with results published in the original report due to fixes in the design and evolution of the modeling tools.

achilleas101, 2024-04-30:
All right, I understand. Thank you again for your guidance!

closed 2024-04-30 by gbarter