IEAWindSystems/IEA-15-240-RWT #205: How to make Cp_Ct_Cq.txt

YBLee02, 2024-05-13:
I made Rotor performance data file from CCBlade using IEA-15-240-RWT-UMaineSemi_ROSCO.yaml included toIEA-15-240-RWT Folder.
But the data i made is not matched to original Cp_Ct_Cq.IEA15MW.txt.

I wonder how can i make Cp_Ct_Cq.IEA15MW.txt same as Original file.

1. Original Cp_Ct_Cq.IEA15MW.txt
[image]

2. New Cp_Ct_Cq.IEA15MW.txt
[image]

### Current behavior
I just revised path in the .yaml.

### Code versions
- Python 3.11
- rosco 2.9.0
- wisdem 3.15.1

ptrbortolotti, 2024-05-13:
Hello [YBLee02](https://github.com/YBLee02), I've tried to replicate the table on GitHub and the one you've posted, and I could not replicate any of the two. The differences probably come from different versions of WISDEM (I'm on 3.15.3). The table is used by ROSCO "only" to tune the gains of the pitch controller, so small differences in the table are not going to change the dynamic of ROSCO all that much. You should be good to push ahead, but please let us know if you get stuck

YBLee02, 2024-05-14:
Hi, ptrbortolotti.
Thank you for your reply.

I have confirmed that Cp,Ct,Cq.txt vary depending on the WISDEM version.

In the future, I will test the rotor performance of Cp, Ct, and Cq.txt, which depends on the WISDEM version.
Your comment helped me a lot.

Best regards.

closed 2024-05-14 by YBLee02