IEAWindSystems/IEA-15-240-RWT #207: Calculation if the 6x6 stiffness and mass matrixes of IEA-15MW blade for Beamdyn

winder365, 2024-05-29:
Dear @ptrbortolotti,

Sorry to disturb you again. The old post is closed (https://github.com/IEAWindTask37/IEA-15-240-RWT/issues/179) , so I open a new post here.

After your adjustment of the code, the 6x6 stiffness and mass matrixes I calculated is more closer to that provide in the IEA-15MW repository (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat). However, there still difference when we use this data to calculated the performance of the rotor.

The power I obtained with the 6x6 matrixes I calculated by SONATA is still about 4~6% lower than the power calculated with the 6x6 matrixes provided in the IEA-15MW repository (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat).

As you mentioned before, the data provided in the IEA-15MW repository was also obtained through SONATA. What confuses me is that, why I cannot calculated the same sectional data as that provided in the IEA-15MW repository through SONATA? 

Does the composite lay-up of the IEA-15MW blade have been optimized and now it is different with its original design? I also try to calculated the 6x6 matrixes  with the lay-up information provided in older version, such as version v1.0. However, the data I obtained still different from data provided in the IEA-15MW repository.

Due to I want to conduct some lay-up design of the IEA-15MW blade, so it is better to reproduce the data same as that provided in the IEA-15MW repository first as the basis.

Looking forward to your help.
Thank for your attention!

winder365, 2024-06-15:
Dear @ptrbortolotti,

Hello, sorry for disturbing, is there any solution?

I have look through the IEA-15MW lay-up materials data provided in here (https://gitlab.lrz.de/HTMWTUM/SONATA/-/blob/master/examples/2_wind_turbines/0_ont5_IEA-15-240-RWT_TipShape_V3_mat.yaml?ref_type=heads) are different from that provided in IEA-15-240-RWT repository (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT.yaml), and the 6*6 stiffness matrixes calculated by these data through SONATA are therefore different.

So does it means the lay-up design or the lay-up materials may be have be adjusted in different release versions? \

Looking forward to your help.
Thank for your attention!

ptrbortolotti, 2024-06-19:
Hello @winder365,
I'm sorry it took me a while to get back to this.
I've now verified that the latest version of [SONATA](https://github.com/ptrbortolotti/SONATA) is generating a somewhat different set of 6x6 K and I matrices, with the biggest differences in the K44 and K66 elements along blade span. This returns a different elastic behavior of the blade, and you are correct, the power does look different. 
Note that @KIDEVLIN and I are actively working on improving SONATA, with an open pull request [here](https://github.com/ptrbortolotti/SONATA/pull/8). In other words, SONATA will change again in the coming weeks and months, and so will (most likely) the matrices coming out of SONATA. (The solver [ANBA4](https://github.com/ANBA4/anba4) is static, but the meshing generated by SONATA is not)

I believe the problem that you're hitting hides in the fact that the [ROSCO controller](https://github.com/NREL/ROSCO) for the IEA15 has never been tuned for BeamDyn. A better job was performed for the more recent [IEA 22 MW](https://github.com/IEAWindTask37/IEA-22-280-RWT)
When the elastic response of the blade changes, ROSCO should be retuned, especially in the pitch behavior.
If you want to want to go down this path, the documentation and the examples of ROSCO are a great starting point.
I hope this helps. Please let us know if you have any further doubt.
Best regards

winder365, 2024-06-20:
Dear @ptrbortolotti 

Thank you for your reply and your effort for the development of SONATA.

I still  have a questions to confirm with you:

 If I want to change the ply-angle of the materials such as the Carbon UD of the Spar cap to obtain the bend-twist coupling, can it be realized by changing the "fiber_orientation" in the .yaml file?

Thank you for your help.

ptrbortolotti, 2024-06-20:
@KIDEVLIN and I have just checked and everything should be hooked correctly. please note that there was some confusion between rad and degrees, which we've just fixed. do a `git pull` in SONATA before moving ahead. in the input yaml file, your fiber angles will need to be in radians

winder365, 2024-06-20:
Dear @ptrbortolotti 

Thank you for your effort, good job!

I have git pull of the new version SONATA, and I have see that you have fixed some confusion as you said.

I change the the orientation of the Carbon UD on the spar cap from 0 deg to 15 deg (namely 0.2618 rad) in the .yaml file and calculate the blade section matrixes.  However, I found that the result is same as what I calculated several days ago. That is to say, the bug you fixed in the latest version may have NO effect on the calculation result？

I am confused with the result, I don't know whether I set the parameters in a proper way.

Thank you for your attention and looking forward to you help.

ptrbortolotti, 2024-06-21:
Oh I'm sorry about that, I'll have to debug what's going on... I'll admit that I've never modeled bend twist coupling in SONATA (I know it works in ANBA4)

KIDEVLIN, 2024-06-21:
Hello @winder365 
I looked into this and tried to reproduce the issue. When I changed the "fiber_orientation" in the Carbon UD on the spar cap 15 deg, I did see impacts on the stiffness matrix compared to the 0 deg orientation. I may be misunderstanding the question, otherwise you should check that the "fiber_orientation" is changed for all grid locations. Let me know if this helps.

winder365, 2024-06-27:
Hello @KIDEVLIN @ptrbortolotti, 

You are right. After changing the "fiber_orientation" in the Carbon UD on the spar cap 15 deg,  impacts on the stiffness matrix compared to the 0 deg orientation can be observed. 
Pro. @ptrbortolotti  have fixed several bugs several days ago. What I mean is the result calculated by the new version of the SONATA is same as that calculated by the old version. Namely the bugs Pro. @ptrbortolotti fixed may has no effect on the simulation result. 
May be I have know the reason. The bugs Pro. @ptrbortolotti fixed are about some confusion between rad and degrees in the webs, so this don't affect the results after changing the  "fiber_orientation" in the Carbon UD on the spar cap.

**Now issue:**
I have an other issues, when I change the "mesh_resolution = 400" to "mesh_resolution = 500" in the "1_sonata_IEA15.py", the program can not be run. May be there are bugs about the mesh resolution.

Thank you for your attention.

Best wishes,

Best wishes, thank for your attention.

KIDEVLIN, 2024-06-27:
Hello @winder365 
It looks like you found the change with the fiber orientation. The issue with the increased mesh_resolution has been fixed in the SONATAmini branch [here](https://github.com/KIDEVLIN/SONATA/tree/SONATAmini) which is currently being worked on but should give better results than SONATA.

winder365, 2024-06-28:
Dear @KIDEVLIN 

Thank you for your effort.

I have try the SONATAmini version, and the bug related to the mesh_resolution  was fixed.  Good job!

However I still confused about the issue I mentioned above. That is the 6x6 stiffness and mass matrixes calculated by SONATA is different from that provide in the IEA-15MW repository (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat).  And as indicated by Pro. @ptrbortolotti , the biggest differences can be observed in the K44 and K66 elements along blade span. 

As you know, k66 is actually the tortional stiffness, and the k66 calculated by SONATA is somehow smaller than that provide in the IEA-15MW repository. So we can think that the SONATA underestimate the tortional stiffness. As the tortional is closely related to the aeroelastic stability of the blade, so it is important. What is more important is that, after changing the ply angle of the spar cab, the bend-twist coupling effect will further decrease the tortional stiffness, and the aeroelastic **instability** may  occur even at the rated point (namely 10.59 m/s, 7.55rpm) of the IEA-15MW wind turbine. So correctly predicted the K66 is of great significance.

Do you found the reason about why the SONATA underestimate the K66 and K44 elements along blade span these days?

Thank you for your help and attention!

Best wishes.

ptrbortolotti, 2024-07-01:
hello @winder365,
NREL originally designed the blade of the IEA15 using https://github.com/WISDEM/WISDEM, which leverages the cross sectional PreComp. Once the design was finalized, a former colleague and I took SONATA and ran it, generating the BeamDyn file that is now on GitHub. In this [paper](https://iopscience.iop.org/article/10.1088/1742-6596/2767/2/022018/meta), we discuss that the rotor as modeled on GitHub is aeroelastically stable when modeled with unsteady aerodynamics, but it is torsionally unstable when modeled with quasi steady aerodynamics. Now over time, and more recently especially thanks to the hard work of @KIDEVLIN, SONATA has been improving, and as a result the K and I matrices generated by SONATA are changing. Necessarily, the aeroelastic results will change. We are not planning to update the BeamDyn files for now and we have not yet checked how the new files impact the aeroelastic stability of the rotor. @KIDEVLIN plans to do this study soon, but only after he is done fixing problems in SONATAmini.
Going back to your question, the new SONATAmini has been improving several weaknesses in the generation of the mesh, and that's where the differences come from.
@KIDEVLIN and I hope that this message helps you in your research!
Regards

closed 2024-07-11 by gbarter