IEAWindSystems/IEA-15-240-RWT #211: IEA 15MW CAD of Generator Rotor and Stator

Tetsuya2000, 2024-06-25:
I was wondering whether the PM DD generator structure car model is available anywhere , that is shared for researchers. 

Please guide me there if that exist 

Thank you

gbarter, 2024-06-25:
There is a CAD directory in the repository:
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/CAD

Tetsuya2000, 2024-06-27:
Hi gbarter, 

Thank you for your previous reply, I found some SLDPRT in deprecated directory. However most are just shell , without windings on the stator slot. 

Are there any complete 15MW CAD model with windings attached? 

I am not very experienced with Solidworks , my main analysis is actually on Comsol, was hoping to find the .STEP or .SLDPRT for each and then analysis the assembly, in sector.

gbarter, 2024-06-27:
Unfortunately, what you see is all that there is.  I don't have any other CAD models not on this repository.

closed 2024-07-11 by gbarter