IEAWindSystems/IEA-15-240-RWT #218: IEA 15 MW design load cases

marwa023, 2024-08-04:
## Description
_I model IEA 15 MW on monopile support under different design load cases. I used Open Fast v3.5.1, I used ROSCO 2.8. My question is why there is no mention in the IEA 15 MW manual for design load cases with faults like 2.x or 7.x or emergency shutdown?
When I tried to simulate DLC 6.2 with yaw errors -180 to 180, I had instabilities between angles +- (20 to 60 ),  I read the same problem in (Loads Analysis of a Floating Offshore Wind Turbine Using Fully Coupled Simulation by J.M. Jonkman and M.L. Buhl, Jr.). when I tried to apply the brake, it didn't change anything (maybe I did something wrong).
But when I tried the solution mentioned in DTU 10 MW manual, which is changing vibration direction to improve the aerodynamic damping by considering the pitch sequence as 70-90-90 or 90-70-90 or 90-90-70, it efficiently removed the instability.  

### My questions are:
1- I understand the solution of changing the pitch sequence to improve the aerodynamic damping because at idling conditions it is minimum, but why specifically 70, not 80 or 60, etc, is there a specific reason? or why the recommended position for blades in parked is feathered not 90,70,90?
2- DLC 6.2 combined extreme wind and wave with a return period 50 years, but the lifetime of the turbine itself of 20 to 30 years, on the same time there are critical yaw errors that give instabilities are between +-(20 to 60) and the remaining yaw errors +-(80-180) results in minimum loads compared to normal operation conditions DLC1.X, so should I excluded DLC 6.2 from the calculations from the beginning 
3-The solution of (bypass it by choosing yaw errors that don’t result in the instability) is it practical? I mean when designing monopile of turbine?
4- when I calculated the out of plane blade tip deflection it was= 36.8 m! (including PSF of 1.35) from DLC 1.4-ECD+r  but in the report it was 22.8 m and the tower clearance is 30 m. what made my calculations incorrect? 

Thanks in advance and sorry if my questions are basic

ptrbortolotti, 2024-08-05:
Hello, thanks for your questions. Some comments:
1) The aeroelastic instabilities during idling/parked conditions are well known. They occur when angles of attack are outside of the verified/validated linear regime and are not to be trusted as conventional numerical models that rely on 2D airfoil polars and 1D beams struggle to reproduce reality, which is characterized by unsteady aerodynamics and 3D effects. This is an active area of research. The solution of pitching one or two blades at an angle different than 90 is somewhat of a workaround, not a rigorous solution.
2) DLC 6.2 is included in the IEC 61400-1 standards. Please refer to that document
3) There is no good answer here. Again, this is an active area of research. @mayankchetan can provide more help here
4) I believe that 22.8m was the deflection estimated by WISDEM, which assumes some steady-state loading. We did not design the rotor of the IEA15 against DLC1.4. Note however that the turbine response in DLC1.4 is highly dependent on the performance of the controller, and the ROSCO controller has not been tuned to perform optimally in DLC1.4. Actually, in my experience is more like the opposite. Unless you tune the shutdown logic, ROSCO will keep the turbine online under extreme yaw misalignments, and the rotor blades will experience huge deflections. That task was part of a broader study on flexible rotors that is described in https://wes.copernicus.org/articles/6/1277/2021/

I hope this helps!

marwa023, 2024-08-05:
Thanks for your quick response and for sharing the information!

That's why DLC 1.4 from IEC 61400-1 standards was the extreme event for the tower base moment and forces, blade root moment, out-of-plane blade tip deflection in my calculations which enables ROSCO during ECD.

1- regarding point#4, as I understand from a quick reading of the reference you provided, to run DLC 1.4 from IEC 61400-1 standards effectively, I have to change the source code of ROSCO and provide a closed loop response characteristics of the rotor for shutdown and recompile it? is this shutdown trigger publicly available or I have to write it?

2- another question please, Does ROSCO support emergency shutdown DLC 5.1 and fault case  DLC 2.X from IEC 61400-1 standards or not, I read before that DISCON for NREL 5MW doesn't support emergency shut down DLC 5.1 IEC 61400-1 standards.

3- In my calculations, for tower top displacement in for aft and side to-side, the governing load cases were 2.x and 5.x. I calculated the displacement from all DLCs and then weighted them by multiplying the displacement* partial safety factor specified in IEC 61400-1 standards for every load case. is that a correct approach or not?

Regards,
Marwa

marwa023, 2024-08-05:
Thanks for your quick reply, I appreciate your help.

marwa023, 2024-08-06:
Hello again, Sorry for asking a lot of questions.

1- For DLC 5.x from IEC 61400-1 standards, Can I simulate emergency shut down at specific time by using ServoDyn input file instead of manipulating the input signal to the shutdown ROSCO controller? For example,  I simulate shutdown at 200 sec by changing ServoDyn as 

PCMode=VSContrl=5
TPitManS =230 sec
PitManRat=8
BlPitchF=90
GenTiStp=GenTiStr=true
TimGenOn=0
TimGenOf=230

Is that an acceptable approach or not?

2-The same thing for DLC 2.1 from IEC 61400-1 standards, I used the approach from (Loads Analysis of a Floating Offshore Wind Turbine Using Fully Coupled Simulation by J.M. Jonkman and M.L. Buhl, Jr.)."For DLC 2.1, we simulated a fault in the collective blade-pitch control system where one blade ignores its command and runs away to the minimum set point of 0º at the full pitch rate of 8º/s. We assume that the turbine’s protection system detects this fault in this situation by simulating a shut down of the turbine; the shut down is initiated after a 0.2-s delay (to account for the time it takes the protection system to detect the fault and take action) by feathering the other two blades at full pitch rate to the maximum pitch setting of 90º."
PCMode=VSContrl=5
TPitManS(1) =60 sec
TPitManS(1) =60.2 sec
TPitManS(1) =60.2 sec
PitManRat=8
BlPitchF(1)=0
BlPitchF(2)=90
BlPitchF(3)=90
GenTiStp=GenTiStr=true
TimGenOn=0
TimGenOf=99999.9

Can I use the same approach with ROSCO or not?

Thanks for your consideration

Regards,
Marwa

ptrbortolotti, 2024-08-07:
@abhineet-gupta can you please help here?

abhineet-gupta, 2024-08-07:
Hi @marwa023 
Yes, Servodyn overrides like TPitManS, TimGenOf etc can be used with ROSCO and correctly override it when activated.

1) Yes, that should be the right approach.
2) Yes, seems like you have the right approach. Not sure if the paper suggests that generator needs to be switched off at 60.2 seconds too or not but that can be achieved by TimGenOf = 60.2 as well.

Abhineet

marwa023, 2024-08-07:
Thanks @ptrbortolotti  and @abhineet-gupta for your help and response.

marwa023, 2024-09-05:
Dear Dr. @ptrbortolotti 

May I ask you a question please?
 In WES paper you shared " An additional shutdown trigger has been implemented, where the shutdown procedure is initiated when the filtered signal of the yaw error is higher than 120◦ in average wind speeds greater than 5 m s−1. This condition is not common in operational design load cases, but it helps to limit the ultimate loads generated during the occurrence of extreme changes of wind direction."
1-why did you use specifically 120 degree?
2- in my simulations of IEA 15 MW under DLC 1.4 from IEC 61400-3, Can I jus use your shutdown logic you shared and implimint it in ROSCO and recomiple it? or I have to change the yaw error to my case 57 degree and the wind speed to ( rated speed +2 ) as in my case this is the crirical case?

Regards,
Marwa

marwa023, 2024-09-06:
Thanks, Dr. @ptrbortolotti  for your detailed reply.
1- Is there a source or paper that explains the variables in the source code to not change them blindly?
2- in WES paper " When the shutdown is triggered, the desired pitch angle is set to 90◦ and the pitch systems are actuated at the maximum rate, whereas the torque actuator attempts to slow the rotor speed to zero in a 30 s time window. Notably, the torque actuator does not necessarily saturate, **which can cause higher loads than the softer slowing of the rotor that has been implemented.** "
in my simulation for IEA 22 under DLC 2.1 according to IEC 61400-3 I activated Servodyn overrides, like what I mentioned before in this post, an**d I had from this case the the extreme event for the tower base moment and force**s. Is my results correct or not please?
[image]
[image]
Regards,
Marwa

closed 2024-11-21 by ptrbortolotti