IEAWindSystems/IEA-15-240-RWT #219: Visualization of IEA 15MW reference FOWT

wonny0503, 2024-08-08:
Dear Communities

I am trying to animate IEA 15MW reference FOWT using Paraview (v5.13.0)  but VolturnUS-S doesn't appear.

I found that visualization is now supported since OpenFAST v3.5.1 (https://github.com/OpenFAST/openfast/pull/1768).

However, I am using OpenFAST v3.5.3 so I don't know what is the problem is. 

FYI, I set WrVTK = 2, VTK_type = 1, and VTK_fields =True.

[image]

May I ask for small tips to solve this problem?

Best regards,
Sangwon

jjonkman, 2024-08-08:
Dear @wonny0503,

The OpenFAST model of the VolturnUS-S makes use of a HydroDyn model with a potential-flow-only solution, with viscous drag accounted for through quadratic drag matrix rather than through the strip-theory solution.  In OpenFAST v3.5.1 and newer, visualization of the floater is available via the strip-theory solution (Morison members).  So, it will not be possible to visualize the floater of this model unless Morison-members are added.  (Note that OpenFAST v3.5.1 and newer does not import the WAMIT _*.gdf_ file to visualize the floater from the potential-flow solution.)

Best regards,

wonny0503, 2024-08-09:
Dear @jjonkman 

I appreciate your detailed and kind reply.

Have a good one.

Best regards,
Sangwon

closed 2024-08-09 by wonny0503