IEAWindSystems/IEA-15-240-RWT #220: cannot find the ROSCO input file for IEA-15-240-RWT

Fede-274, 2024-08-19:
## Description
I cannot find the file "libdiscon.dll" and "DISCON.IN" that are requested to use the ROSCO controller alongside Fast. Therefore I cannot complete the simulation with Fast. Can someone send me these files or tell me where to find them?

marwa023, 2024-08-22:
Hello,
Can anyone help me to find DTU Wind Energy controller (DLL, and input file) that can be run with OpenFast for turbines IEA 15 MW and IEA 22 MW?

Regards,
Marwa

jennirinker, 2024-08-26:
Hi @marwa023,

Since the comment (and request) is different than the original issue, I have made a new issue corresponding to this request: #221. I will follow up in that thread.

@gbarter Since it looks like you responded to the original request, I will mark this Issue as closed. :)

@Fede-274: If Garrett's comment did not solve your issue, please comment here and we can reopen it.

Cheers,
Jenni

closed 2024-08-26 by jennirinker