IEAWindSystems/IEA-15-240-RWT #223: Tower and  Rotor nacelle assembly mass

lk73438702, 2024-09-22:
Hi everyone:
I used the openfast linearization function to calculate the mass matrix of the IEA-15MW. The mass matrix I obtained is different from the mass in the Definition of the IEA 15-Megawatt Offshore Reference Wind. The following picture is the calculation result.In the report, Tower mass = 860t, Blade mass = 65t, but the calculated result is Tower mass = 1466.5t, Blade mass = 68.5t.
[image]
[image]
Best regards,

gbarter, 2024-09-22:
The design has evolved a bit since the original report release.  Please review the release notes and the FAQ and let me know if your doubts persist.

lk73438702, 2024-09-23:
Dear @gbarter ，
The data in the above picture is calculated using OpenFast 3.5. Can you tell me if the result is correct? Thank you very much。
Best regards，

gbarter, 2024-09-23:
I'm not sure what the question or worry is.  OpenFAST has mass properties listed directly in the ElastoDyn file and in the blade/tower mass density per length inputs.  If you are using the latest files offered here, then you are achieving the correct mass values.

You can also double check things against the Excel file in the [Documentation](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/Documentation) folder.