IEAWindSystems/IEA-15-240-RWT #224: Computation of natural frequencies of tower

MassimoSirigu, 2024-09-23:
Good afternoon,
I have a question about how natural frequencies of the floating tower is computed.
I'm doing a modal analysis of the floater+turbine is Ansys mechanical using shell elements.
I compared two cases:
1) model including tower only and RNA and Platform mass and inertia modeled as point masses (including added mass).
2) a complete model including also a shell element of the platform (added mass, internal ballast, internal girders and stiffeners included).
Despite the total mass and inertia of the platforms are the same, the natural frequencies between the two models are different:
1) about 0.45 Hz
2) about 0.34 Hz
Based on your team's experience, do you confirm that the stiffness of the platform itself is so important for modal analysis? 
did you perform the optimisation of the floating tower using model number 1 or you used a different model to compute natural frequencies of tower?

Thank you

gbarter, 2024-09-24:
Hello.  I don't quite follow points 3 or 4 in your list.  We use roughly the approach you described in point 1, although the key modeling aspects to get right are the added hydrodynamic mass of the platform and the floating boundary condition of the platform.  If you use a different boundary condition, then you are likely to get the wrong answer.

We generally use WISDEM (with Frame3DD doing the structural analysis) or OpenFAST (I assume HAWC2 would work too) to estimate floating tower natural frequencies.  I have not used Ansys to do the same, although I presume that is something that it could accomplish.

MassimoSirigu, 2024-10-01:
Hi, thanks for your answer. I clarify for you and future readers, for point 1 I used a free-free boundary conditions and point masses with Inertia at both ends (tower modeled with beam elements in Ansys). The frequency of 1st mode (excluding first 6 modes of rigid body motion) was about 0.45 hz.
Second model is a complete shell element model also in Ansys, and gave me 0.39 Hz ((excluding first 6 modes of rigid body motion).