IEAWindSystems/IEA-15-240-RWT #225: How to output the wave height at a specified location.

banzhiqing, 2024-09-24:
Dear Professor,

I have encountered an input option in the file named IEA-15-240-RWT-UMaineSemi_HydroDyn.dat, which refers to "the number of points where the incident wave elevations can be computed." May I inquire if this option enables the output of wave elevations at user-specified locations? For instance, could it provide wave elevation data at a fixed point 10 meters in front of the wind turbine (at coordinates x = -10, y = 0)? If it is possible to output wave elevations at a specified location, what would be the procedure to configure it? Your guidance on this matter would be greatly appreciated.

Best regards
[image]

jjonkman, 2024-09-25:
Dear @banzhiqing,

Yes, that is correct.  Specify `NWaveElev` to the number of points (up to 9) that you want the wave elevations output at and specify the (X,Y) locations of those points in the list of `WaveElevxi` and `WaveElevyi`.  Then output the wave elevation at those points by listing them in the HydroDyn `OutList`, i.e., `Wave1Elev` for point 1, `Wave2Elev` for point 2, etc. up to `NWaveElev`.  If all you want to do is output the wave elevation at (-10,0) m, keep everything as you have except set `WaveElevxi` = -10 m, in which case `Wave1Elev` would be the wave elevation at this point.

Best regards,

banzhiqing, 2024-09-25:
> Dear @banzhiqing,
> 
> Yes, that is correct. Specify to the number of points (up to 9) that you want the wave elevations output at and specify the (X,Y) locations of those points in the list of and . Then output the wave elevation at those points by listing them in the HydroDyn , i.e., for point 1, for point 2, etc. up to . If all you want to do is output the wave elevation at (-10,0) m, keep everything as you have except set = -10 m, in which case would be the wave elevation at this point.`NWaveElev``WaveElevxi``WaveElevyi``OutList``Wave1Elev``Wave2Elev``NWaveElev``WaveElevxi``Wave1Elev`
> 
> Best regards,

Dear @jjonkman,

Thank you very much for your assistance, I now understand how to set these parameters.

Best regards,

banzhiqing, 2024-10-18:
> Dear @banzhiqing,
>
> Yes, that is correct. Specify `NWaveElev` to the number of points (up to 9) that you want the wave elevations output at and specify the (X,Y) locations of those points in the list [...]

 Dear @jjonkman,

I would like to ask a question regarding the use of the JONSWAP spectrum for wave generation. How many frequencies (or phases) are there in total within the range [WvLowCOff, WvHiCOff]?  Additionally, if I wish to modify this quantity, where can I make the adjustment?

Best regards,
[image]

jjonkman, 2024-10-18:
Dear @banzhiqing,

The frequency step used by HydroDyn is related to `WaveTMax`, that is, WaveDf = 1/`WaveTMax` in Hz, or WaveDomega = 2*pi/`WaveTMax` in rad/s.

Best regards,

banzhiqing, 2024-10-18:
> Dear @banzhiqing,
> 
> The frequency step used by HydroDyn is related to `WaveTMax`, that is, WaveDf = 1/`WaveTMax` in Hz, or WaveDomega = 2*pi/`WaveTMax` in rad/s.
> 
> Best regards,

Dear @jjonkman,

Thank you very much for your help! I now understand this parameter.

Best regards,

banzhiqing, 2024-10-19:
> Dear @banzhiqing,
> 
> The frequency step used by HydroDyn is related to `WaveTMax`, that is, WaveDf = 1/`WaveTMax` in Hz, or WaveDomega = 2*pi/`WaveTMax` in rad/s.
> 
> Best regards,

Dear @jjonkman,

I would like to kindly seek your guidance on a question. In the HydroDyn.dat file you provided, the default value for WaveTMax is 850 s. When simulating the motion response of a floating offshore wind turbine (FOWT) under the combined effects of random wind and waves for a total duration of 6000 s (as indicated by TMax in the .fst file), should WaveTMax be set to a value greater than 850 s, or is WaveTMax = 850 s already considered sufficient for representing irregular waves? Could you please advise on the appropriate settings for TMax and WaveTMax in this context?

Best regards,

jjonkman, 2024-10-19:
Dear @banzhiqing,

When modeling irregular sea states (`WaveMod` = 2), our HydroDyn modeling guidance (https://openfast.readthedocs.io/en/main/source/user/hydrodyn/modeling_considerations.html#waves) recommends setting `WaveTMax` = 3600 s, even when `TMax` is less than 1-hour.  I was not the one to make the _IEA-15-240-RWT-UMaineSemi_HydroDyn.dat_ example, but I don't see that this guidance was followed.  But if you are running simulations longer than 1 hour, you could increase `WaveTMax` to `TMax` to ensure that the irregular waves don't repeat within the simulation.

Best regards,

banzhiqing, 2024-10-20:
> Dear @jjonkman,
> 
> Thank you very much for your help! I now understand this parameter.
> 
> Best regards,

Dear @jjonkman,

Thank you very much for your help! 

Best regards,

banzhiqing, 2024-10-20:
> Dear @banzhiqing,
> 
> When modeling irregular sea states ( = 2), our HydroDyn modeling guidance (https://openfast.readthedocs.io/en/main/source/user/hydrodyn/modeling_considerations.html#waves) recommends setting = 3600 s, even when is less than 1-hour. I was not the one to make the _IEA-15-240-RWT-UMaineSemi_HydroDyn.dat_ example, but I don't see that this guidance was followed. But if you are running simulations longer than 1 hour, you could increase to to ensure that the irregular waves don't repeat within the simulation.`WaveMod``WaveTMax``TMax``WaveTMax``TMax`
> 
> Best regards,

Dear @jjonkman,

Thank you very much for your help.

Best regards,

banzhiqing, 2024-10-20:
> Dear @jjonkman,
> 
> Thank you very much for your assistance, I now understand how to set these parameters.
> 
> Best regards,

Dear @jjonkman,

Thank you very much for your assistance, I now understand how to set these parameters.

Best regards,

banzhiqing, 2024-10-20:
> Dear @banzhiqing,
>
> When modeling irregular sea states (`WaveMod` = 2), our HydroDyn modeling guidance [...]

Dear @jjonkman,

Thank you very much for your assistance, I now understand how to set these parameters.

Best regards,