IEAWindSystems/IEA-15-240-RWT #226: CDmax estimation

Zheng-Haojie, 2024-10-07:
Hi!
I am trying to simulate a wind turbine with dirty blades, so I would like to modify the polar curve of airfoils. When I use AirfoilPrep to expand the drag coefficient, I need to estimate CDmax through aspect ratio.  I checked the CDmax of the clean airfoil, it was clearly not calculated based on the actual aspect ratio. For example:
_node 48, chord length=1.94, blade length=117m, aspect ratio is approximately 60.3, and Airfoil estimates a CDmax of 2.01. But the CDmax in the airfoil file is 1.4999_
Therefore, I would like to know how you took the value of CDmax when designing a 15 MW wind turbine model? Is there a recommended value for aspect ratio?
Looking forward to your answer. If I have any misunderstandings, please point them out.

ptrbortolotti, 2024-10-07:
hello @Zheng-Haojie. thanks for your message. I am not sure I follow... those airfoil polars were computed in CFD in the linear regime, and then extrapolated with Viterna, which takes Cd_max as a user defined input. Cd_max=1.5 is a rather standard input. I am not familiar with a relationship between aspect ratio and Cd_max, can you please elaborate about it?

Zheng-Haojie, 2024-10-08:
Thank you for your prompt response！

The Viterna method is also used in AerofoilPrep. And estimate CDmax according to this formula:
aspect ratio = (rotor radius / chord_75% radius)
cdmax = 1.11 + 0.018*AR

Here is the description in its code: 
Extrapolates force coefficients up to +/- 180 degrees using Viterna's method
        :cite:`Viterna1982Theoretical-and`.

        Parameters
        ----------
        cdmax : float
            maximum drag coefficient
        AR : float, optional
            aspect ratio = (rotor radius / chord_75% radius)
            if provided, cdmax is computed from AR
        cdmin: float, optional
            minimum drag coefficient.  used to prevent negative values that can sometimes occur
            with this extrapolation method
        nalpha: int, optional
            number of points to add in each segment of Viterna method

        Returns
        -------
        polar : Polar
            a new Polar object

        Notes
        -----
        If the current polar already supplies data beyond 90 degrees then
        this method cannot be used in its current form and will just return itself.

        If AR is provided, then the maximum drag coefficient is estimated as

        >>> cdmax = 1.11 + 0.018*AR_

However, it seems that this recommended formula has not been adopted in the design of reference wind turbines. When designing the 5MW RWT, aspect ratio adopted as 17, and CDmax=1.416。For 15MW RWT, I observed that the CDmax at different nodes gradually increases from the root to the tip, In the pointed airfoil, it is taken as your suggested value of 1.5.
[image]

It seems that CDmax is usually assumed by humans. I would like to know (1) what ideas did you and your team use when designing the value of CDmax? (2) From the estimated formula, it seems that CDmax is only related to the geometry of the blade, so is CDmax a relatively stable value? Or is it a less important value?
Any response would be beneficial to me, thank you for your development and maintenance!

ptrbortolotti, 2024-10-09:
hello, thank you for the explanation. as you guessed, we have not used that original relationship between aspect ratio of the airfoil and maximum drag coefficient. I guess we could have as opposed to assume a CDmax a priori. 
I would claim however that CDmax is not a critical quantity as 1D lifting lines are known to be deficient when it comes to storm loads and flows hitting at +-90 degrees of angle of attack. so I'd caution relying on variations in CDmax to draw strong conclusions about storm loads. I would refer to this recent study https://doi.org/10.3390/en17174328 for more details about 360 deg polars

Zheng-Haojie, 2024-11-27:
Thank you very much，this research provides a lot of valuable information！