IEAWindSystems/IEA-15-240-RWT #231: Material strength issue

vacupenda, 2024-11-11:
Is the strength data of these materials obtained when the fiber angle is zero degrees?Because I found that the data for each parameter is different in all three directions.The image below shows the material data from the 'IEA-15-240-RWT_VolturnUS-S.yaml'.
[image]

gbarter, 2024-11-12:
These composite layers are anisotropic materials, so some material properties will depend of the orientation of the load relative to the orientation of the fibers.  For this material you highlighted, yes you can assume the dimension1 is aligned with the uni-axis fibers.

closed 2024-11-21 by ptrbortolotti

vacupenda, 2025-02-09:
Furthermore, for the two materials shown in the figure, when I applied the rated load (calculated by OpenFAST) to the blade and used a material mechanics analysis tool to calculate their strength ratios, I found them to be less than 1, which indicates that both materials would fail. Additionally, one paper mentioned, “The biaxial fibers [45/−45] and triaxial fibers [45/02/−45] can be represented by multiple uniaxial fibers in different directions.” Can I use the material “Glass_unin” to represent both “glass_biax” and “glass_triax”? When I apply the ply stacking using the aforementioned method rather than directly using the data shown in the figure, the calculated strength ratios are always greater than 1. However, I am not sure if this method is feasible.
[image]
[image]