IEAWindSystems/IEA-15-240-RWT #232: Inquiry About HydroData for the UMaine VolturnUS-S Model

Jingyun-Kim, 2024-11-13:
## Description
Hello,

I am conducting research on the VolturnUS-S model using OpenFAST and Nemoh.

### Steps to reproduce issue
1. Run simulations for the VolturnUS-S model using Nemoh.
2. Compare the hydrodynamic coefficients (Added mass and Damping coefficients) with WAMIT results.
3. Review WAMIT results from both the 2020 Technical Report and the updated 2022 data on GitHub.

### Current behavior
The WAMIT results presented in the 2020 Technical Report differ from those updated in 2022 on GitHub, leading to confusion about which dataset should be used for comparison with Nemoh results. Additionally, no experimental data for Added mass or Damping coefficients for the VolturnUS-S model is available for validation.

### Expected behavior
Clarification on which WAMIT dataset—2020 Technical Report or 2022 GitHub update—should be used for reliable comparison. Also, guidance on whether experimental data for hydrodynamic coefficients is accessible for this model.

### Code versions
_List versions only if relevant_
- Python
- …

gbarter, 2024-11-13:
Thank you for your interest in the 15-MW reference turbine.  You will find some comments in the FAQ that the original reports and files did have some bugs and over time we have brought the model to be more consistent with the original design intent.  The current model files should be treated as "truth" when differences arise with the original report.

Jingyun-Kim, 2024-11-13:
It seems that my efforts to find the materials were insufficient. Thank you very much for your response.