IEAWindSystems/IEA-15-240-RWT #234: Error in OpenFAST calculation caused by changing the chord length and the material layup.

ZHaaaAO, 2024-11-21:
Dear Professor:

I am currently working on some optimization tasks for the IEA-15MW wind turbine blade, but I have encountered some difficulties that I am unsure how to address.

After modifying the chord length and the material layup of certain sections, I updated the input files accordingly, for example, the BeamDyn_blade.dat and AeroDyn15_blade.dat, etc.

When I calculate the performance of the UMarineSemi wind turbine model using OpenFAST (3.5.0) and ROSCO (2.9.0) with the modified blade structure, the wind turbine becomes unstable under turbulent wind conditions, exhibiting some high-frequency characteristics (Figure 1: The IEA-15MW generator power after the blade structure change ).
[image]

I have also noticed that when I run the program, the command window displays a warning message: 
[image]

What does this mean?

I suspect this issue may be due to the change in aerodynamic performance of the new blade structure.

I tried setting SkewMod = 1 in the AeroDyn15.dat file as the warning suggested, and the calculation became stable.

Could you provide any suggestions or ideas?

Looking forward to your reply!

ptrbortolotti, 2024-11-21:
I am not sure how your new model behaves, but I think this forum post might help https://forums.nrel.gov/t/wake-skewed-angle-model/1432/5

ZHaaaAO, 2024-11-22:
Dear @ptrbortolotti 

**Thank you for your prompt response!** I've been enlightened by the forum discussion on the Wake-skewed model, which has made a significant impact on my understanding.

I would like to share with you the behavior of my new model, as I am puzzled by the phenomena that have emerged. All simulations have been conducted under the UMaine-semi conditions, with all DOFs for the platform, blades, and tower being unrestricted.

Figures 1.1-3 illustrate the performance comparison between two models under the designed DLC conditions, with a constant wind speed of 10.59 m/s. （The blue line represents the new model, while the orange line corresponds to the original model.）
[image]
Figure1.1 Blade pitch angle comparison

[image]
Figure1.2  Generator power

[image]
Figure1.3 Rotor speed

As observed, both models achieve stability after a brief period. However, it appears that the modified model exhibits a greater challenge in reaching convergence.

Figures 2.1-4 depict the performance of the two models under a constant wind speed of 17.2 m/s. The color coding for each model is consistent with that used in the previous figures.

Figure2.1 Blade pitch angle comparison(17.2m/s)
[image]

Figure2.2 Generator Power comparison (17.2m/s)
[image]

Figure2.3 Generator torque comparison(17.2m/s)
[image]

Figure2.4 Rotor speed comparison(17.2m/s)
[image]

The generator torque of the new model appears to stabilize, yet the behavior of the blade pitch angle, generator power, and rotor speed remains perplexing. Notably, the generator torque has been observed to even take on negative values.

This is impossible, right? After post-processing the visual VTK files generated by OpenFAST using ParaView, I observed that the blades were oscillating at a relative-high frequency. This phenomenon could potentially be the source of the anomalous results, casting doubt on the reliability of the simulation outcomes.

So I guess the control strategy is not suitable for the new model, because the structure change has a influnce on Cp, Ct, Cq and stiffness. But after I re-tuned the DISCON.IN using the new blade input file, the result is also not satisfactory. 

**In brief, here are my questions:**
①For a new or optimized blade model, which performance indices are most critical to monitor closely? Should I prioritize generator power, wind turbine stability, and control strategy, or are there other factors I should consider? I would greatly appreciate any suggestions, as this matter is of utmost importance and has caused me considerable confusion.

②I have modified the chord and material layup of certain sections of the blade, which is expected to alter the stiffness and aerodynamic performance of the wind turbine blade. However, I did not adequately consider the continuity of the chord and material distribution. Is the continuity a significant factor, and should I revise my optimization approach?

③I have recalculated the power coefficient (Cp), and I've found that all the results are lower than those of the original model. Does this indicate that the IEA-15MW blade structure has already achieved optimal aerodynamic performance?

Thank you for your patience！

ptrbortolotti, 2024-11-27:
Hello,
I suspect that by changing the turbine design you've now made your design unstable. This paper could be a starting point to look into that https://iopscience.iop.org/article/10.1088/1742-6596/2767/2/022018/meta
This new app can help you investigate things https://github.com/OpenFAST/acdc
Without knowing more, increasing torsional stiffness and edgewise stiffness usually helps mitigate (or eliminate) the instability. Unsteady airfoil aerodynamic (UAMod=4) can also help.
Regards

ZHaaaAO, 2024-12-02:
Dear @ptrbortolotti ,

I wanted to extend my sincere gratitude for the helpful insights and resources you've provided regarding the potential instability in wind turbine design. 

Your suggestion to review the paper and the introduction to the ACDC app are incredibly valuable starting points for my investigation.

Thank you sooooooo much!

Warm regards

closed 2024-12-02 by ZHaaaAO