IEAWindSystems/IEA-15-240-RWT #237: Aerodynamic force in SS direction - IEA 15 MW monopile

Fede-274, 2024-12-05:
I am analyzing the side-to-side (SS) oscillations of the IEA 15 MW monopile turbine. I want to calculate the component of the aerodynamic force at the tower top in the SS direction. 

I am reading the excel file of possible outputs from ElastoDyn, which parameter is right for me? Or are there other parameters that I am not considering at the moment?
[image]

jjonkman, 2024-12-07:
Dear @Fede-274,

The ElastoDyn force outputs you are showing are reaction forces within the shaft, and so, not include aerodynamic applied forces, but also gravity forces, inertia forces, etc.  If you want to isolate the aerodynamic applied forces (as you implied), you can output these from AeroDyn.

AeroDyn outputs `RtAeroFyh` and `RtAeroFyi` (likewise for xh, zh, xi, and zi) are the aerodynamic applied forces integrated across the rotor and output in the hub (moving and rotating with the hub) and inertial frame coordinate systems, respectively.  While not exactly side-side, you could one of these sets of outputs and change the output coordinate system in a post-processing step based on rotations in the system, depending on exactly what you want.

Best regards,

Fede-274, 2024-12-07:
Dear @jjonkman  , thank you for your gentle response,

I have some questions, such as:
- does the output “_LSShftFys_” from ElastoDyn, also include the aerodynamic forces (besides the inertial and gravitational forces you mentioned)? 
- the Ys axis in which “_LSShftFys_” is calculated, is it the same Yp axis in which the SS displacement of the tower top “_YawBrTDyp_” is calculated?
- By “reaction forces within the shaft” you mean the forces that are applied on the shaft?

Finally, I am interested in the forces in SS direction that do not rotate, so I guess I only need to consider “_RtAeroFyi_ ”, but I do not understand the kind of rotation I should do in post processing ... isn't the Yi axis already oriented in the SS direction just like the Yp axis where the SS displacement of the tower top “_YawBrTDyp_” is calculated? If not, what kind of rotation should I make?

Thanks in advance.

jjonkman, 2024-12-09:
Dear @Fede-274,

Here are my responses:

- By "reaction force", I mean the force transmitted through the shaft, which are the applied forces (aerodynamic, mass / gravity) minus inertia forces (mass / acceleration).  For a simple mass-spring-damper+force model, the reaction would be the spring force plus damper force, which equals the applied force minus inertia force.
- `LSShftFys` is the reaction force in the shaft transverse to the shaft along the ys/yn axis.
- ElastoDyn output `LSShftFys` includes aerodynamic applied forces, but it includes other forces (gravity, inertia) as well.
- ys points in the same direction as yn, which only equals yp when the nacelle-yaw angle is zero.  The yi axis only equals the yp axis when the side-side tower-bending deflection is zero and the platform rotations are zero.  These coordinate systems used by ElastoDyn are documented in the Coordinate Systems section old FAST v6 User's Guide: https://openfast.readthedocs.io/en/main/_downloads/d8bd014121d6505cb25cf49bee5eaa80/Old_FAST6_UsersGuide.pdf.
- The rotations will depend on which ElastoDyn degrees of freedom you have enabled and their instantaneous angles (platform, tower-bending, nacelle-yaw).

Best regards,

closed 2025-04-10 by ptrbortolotti