IEAWindSystems/IEA-15-240-RWT #238: IEA-15-240-RWT_tabular.xlsx - Rotor Performance

KuanGordon, 2024-12-11:
[image]

I recently downloaded "IEA-15-240-RWT_tabular.xlsx". I noticed something weird in the "Rotor Performance" table that I'd like to report. 

Specifically, the power and power coefficient have negative values when wind speed is lower than 7 m/s, which seems incorrect. It's possible there may be other weird data in the remaining columns as well, but I'm not sure.

jennirinker, 2024-12-12:
I have seen in some models negative powers at low wind speeds when there is a minimum rotor speed, because there is not enough wind to get the rotor to spin at the required minimum rotor speed.

That is not what's going on here. There is definitely a bug somewhere -- the pitch angles are also super weird.

@gbarter It looks like this bug was introduced in commit https://github.com/IEAWindTask37/IEA-15-240-RWT/commit/f4e0b849db5c3baad499500e18e6533f5333aa24 in April. In the commit before (https://github.com/IEAWindTask37/IEA-15-240-RWT/commit/fee8f9355029f35f9abc345f79df712be4a755db), the Excel sheet is fine. 

@KuanGordon Thank you for bringing this to our attention!

gbarter, 2024-12-12:
Thank you for catching this indeed!  There is one line of negative power at cut-in due to the dynamics @jennirinker describes, but the other lines are fixed now.  Not quite sure how those bugs crept in, but they are cleaned up in #239

closed 2025-04-10 by ptrbortolotti