IEAWindSystems/IEA-15-240-RWT #244: Missing specification of liquid ballast missing in WindIO specification of floating platform

fabiopierella, 2024-12-30:
## Description

In the WindIO definition of the floating platform, I cannot seem to find an indication that the lower pontoons are flooded.
After a conversation with  @gbarter [here](https://github.com/IEAWindTask37/windIO/issues/76), I think the correct way to specify fluid ballast would be to add something like the following block in the specification of the lower pontoons, together with a new material called `ocean_water`.

Let me know what you think about this solution.

Best,
Fabio

```ballasts:
   - variable_flag: False
     material: ocean_water
     volume: 12345
     grid: [0.0, 1.0]
```

dzalkind, 2024-12-30:
I agree that there should be some way to specify whether a member is filled/flooded.  It's been on my TODO list to add this to WindIO/WISDEM.

In my mind, it was just a `flooded: True` flag.  If a specific volume is specified, then it could conflict with the calculated volume that's defined by the member diameter/thickness and joint locations.

gbarter, 2024-12-31:
WindIO improvement work is planned for the next 2-3 months.  Let's keep this Issue open as a placeholder / reminder to add a `flooded` flag.  This will be useful for fixed-bottom monopiles too.