IEAWindSystems/IEA-15-240-RWT #247: Natural frequency of (monopile+tower) IEA-15-240-RWT

hishamtariq19, 2025-01-06:
I am working on optimally tuning a TMD ( two independent, 1 DOF, linear mass spring damping elements) to be incorporated into IEA-15-240-RWT. For this purpose, I require first natural frequency of the support structure (tower+monopile) of the turbine in fore-aft and side-side directions. I used WISDEM and got the following results:
fixedse.fore_aft_freqs = [0.18144674747673337, 0.978611206087042, 2.05944384649349] Hz
fixedse.side_side_freqs = [0.1804896783195183, 0.8925907682140354, 1.9511661029971008] Hz

Are the above-mentioned values of natural frequencies reasonable, as the report "Definition of the IEA Wind 15-Megawatt Offshore Reference Wind Turbine" mentions 0.17 Hz as the first tower monopile frequency?

gbarter, 2025-01-06:
The model has evolved since the report, so values and loads/performance stated in the report shouldn't be taken as the firm truth.  You can see various comments to this effect in the FAQ.  So, yes, you can interpret the 0.18Hz values as more accurate, although the differences compared to the report value of 0.17 are within the likely margin of error.

In case it is helpful, there are some [examples](https://github.com/WISDEM/WEIS/tree/main/examples/06_IEA-15-240-RWT) in the WEIS project with regard to TMDs for the 15-MW work based on prior projects.  Please reach out via email if you would like to know more there.

hishamtariq19, 2025-01-08:
Thank you very much for the response. I realized some changes in the design over the years. Thanks for the clarfication. Although I have one more question on modes.
From WISDEM, I requested "fixedse.tower_fore_aft_modes" and  "fixedse.tower_side_side_modes" to obtain the coefficients and compare them with OpenFAST

WISDEM Output :
fixedse.tower Fore-aft mode shapes = [[  1.03236664   0.34772544  -1.09981178   1.00583855  -0.28611886]
 [  2.5900872    0.34971887  -3.77103959   2.53271351  -0.70147998]
 [-14.91925114   3.52805392  31.0950484  -22.11404063   3.41018945]]
fixed.tower Side-side mode shapes = [[  1.02949883   0.34772587  -1.095425     1.00262629  -0.28442599]
 [  2.31798533   0.35618082  -3.26291027   2.27260931  -0.68386519]
 [-47.82592487   8.47639446  96.01436916 -65.38808001   9.72324126]]

When compared to OpenFAST, there are some variations. I tried to plot 2nd mode in fore-aft direction as depicted below:

[image]

Can you suggest what can be changed in WISDEM to achieve a good match with mode shapes of OpenFAST? I also intend to resize the support structure and assess the re-design in OpenFAST. Hence, it is crucial that I obtain correct mode shape  coefficients from WISDEM.

Finally, thank you for referring to the WEIS project. As of now, I am using a simplified tuning method for TMD (installed at the top of the tower), but I will look into WEIS in the future to compare the two tuning approaches. I will certainly contact you through email.

gbarter, 2025-01-08:
No doubt, the ElastoDyn tower mode shape coefficients have been a fragile element of the OpenFAST IEA 15MW model.  When I run things locally though, my numbers don't match yours for the second mode coefficients.  I see in my output:
```
fixedse.tower_fore_aft_modes,output,,"[
[1.0363262228783998, 0.14407995823324998, -0.6073010265902048, 0.6046724191912101, -0.1777775737126554], 
[3.6622372539770134, -0.6339120611328884, -3.3516698212231235, 1.8562095296784118, -0.5328649012994127], 
[-2.016158735345492, 1.8286738249579435, 3.383383202513609, -2.3509299501782133, 0.15503165805215274]]",

fixedse.tower_side_side_modes,output,,"[
[1.0341478332576524, 0.14466631900138802, -0.6053976870489439, 0.6033802503171184, -0.1767967155272152], 
[3.1007051334896465, -0.4501949227813007, -2.71721930407064, 1.5892638458138435, -0.5225547524515491], 
[-2.4576439052284504, 1.9421138559227649, 3.8783885749523197, -2.475493875491773, 0.11263534984513876]]",
```

This does match what is in the current OpenFAST model [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-Monopile/IEA-15-240-RWT-Monopile_ElastoDyn_tower.dat).  Perhaps double check that you are using the latest versions of both WISDEM and the IEA 15MW model?

hishamtariq19, 2025-01-09:
I used the IEA 15 MW file from [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT.yaml) .
I uninstalled and reinstalled WISDEM. I get the same output as mentioned in my above comment, and the numbers don't match yours.
Can you advise me what I can do in this regard?