IEAWindSystems/IEA-15-240-RWT #248: Missing elastic_properties_mb for Hub & Nacelle

TheMercer, 2025-02-04:
WindIO hub and nacelle are missing section "elastic_properties_mb", same issue with older models also.

ptrbortolotti, 2025-02-04:
You are correct, we're actively working on this gap here https://github.com/IEAWindSystems/windIO/pull/87
In the meanwhile, please refer to the [OpenFAST](https://github.com/IEAWindSystems/IEA-15-240-RWT/tree/master/OpenFAST) or [HAWC2](https://github.com/IEAWindSystems/IEA-15-240-RWT/tree/master/HAWC2) models

TheMercer, 2025-02-04:
Thank you, for the prompt reply and I will look ta WindIO#87 now. 
I will see for the meantime dig up the values from the other models like you suggested

TheMercer, 2025-03-05:
@ptrbortolotti hope your doing well. 

Just wondering, when this PR is done, will the WindIO file include the extra information missing or will it just have the fields empty ?

ptrbortolotti, 2025-03-05:
we'll do our best to populate those fields once windIO v2.0 is finalized

TheMercer, 2025-03-05:
Thank you, if I can help in anything please reach out. I don't mind lending a hand

TheMercer, 2025-07-11:
Oh brilliant. I will check and wait for final version. I will also need to update my scripts to WindIO V2 for converting to orcaflex haha at somepoint

TheMercer, 2025-07-11:
I guess this is after just what's time like for WEIS/Wisden being updated to work with V2.0

TheMercer, 2025-07-11:
I will say, seeing the conversion happen over the months alot has changed and i do think it's nice now. 

Old version had a lot of legacy stuff