IEAWindSystems/IEA-15-240-RWT #249: Blade Layup Material Issue

vacupenda, 2025-02-09:
For the two materials shown in the figure, when I applied the rated load (calculated by OpenFAST) to the blade and used a material mechanics analysis tool to calculate their strength ratios, I found them to be less than 1, indicating that both materials would fail. Additionally, a paper mentioned([The Impact of Bend–Twist Coupling on Structural Characteristics and Flutter Limit of Ultra-Long Flexible Wind Turbine Composite Blades](https://www.mdpi.com/1996-1073/16/15/5829)), “The biaxial fibers [45/−45] and triaxial fibers [45/02/−45] can be represented by multiple uniaxial fibers in different directions.” Can I use the material “Glass_unin” to represent both “glass_biax” and “glass_triax”? When I perform the ply stacking using the aforementioned method rather than directly using the data shown in the figure, the calculated strength ratios are always greater than 1. However, I am not sure if this method is feasible.

[image]

[image]

[image]

ptrbortolotti, 2025-02-09:
hi @vacupenda, the nominal blade did not undergo a detailed structural design, and so it is possible that some stress limits are exceeded. one word of caution is maybe around loads: in addition to material strength values, pay also attention to what loads you are using to compute the stresses.
Note that in the [README](https://github.com/IEAWindSystems/IEA-15-240-RWT/tree/master) you are going to see a couple of studies listed where a more rigorous structural design was performed. I would look into those
as for laminate theory, a triax can be seen as a combination of uniax and biax, but it's hard to say that it will be a 1-1 match. Triax fabrics have fibers at 45/0/-45 directions in specific percentages of volume, so you cannot just combine two random uniax and biax and you will get the properties of the triax

vacupenda, 2025-02-10:
Thank you for your response. I found all the materials and layups in this file(https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT.yaml]. I used VABS to calculate stiffness and strength ratios. The stiffness results match correctly, but when calculating strength (with loads obtained from OpenFAST under rated conditions, providing forces and moments in three directions at different spanwise positions), the strength ratios for “glass_biax” and “glass_triax” are particularly low, while all other materials are normal. Therefore, I replaced these two materials with unidirectional fibers and applied layups of 45/-45 and 45/0/-45, respectively. I'm not sure if the issue originates from these two materials.Are glass_biax and glass_triax material properties that have already been mixed using unidirectional glass fibers? If I were to create the layup using unidirectional glass fibers myself, what proportions should I use for different fiber angles?

ptrbortolotti, 2025-02-10:
The material properties are discussed in this article https://doi.org/10.5194/wes-7-19-2022, see Table 1. The values should be accurate and come from the open-source Montana State University composite database

I'm sorry, I don't have an immediate answer about the proportion of UD vs BX fibers in a TX laminate, I am sure commercially you will find different TX with different proportions

vacupenda, 2025-02-11:
Thank you for your response. This literature has been very helpful to me. Additionally, when I use BeamDyn to calculate the loads at different spanwise positions of the blade, I need to input a BeamDyn.dat file, which contains the stiffness matrices at different spanwise positions. Currently, I am using VABS to calculate this stiffness matrix. However, when using VABS for the calculation, translating the cross-section to different reference points results in different stiffness matrices. Therefore, in order to obtain the stiffness matrix as required in the file, at which reference point should I define the cross-section?

ptrbortolotti, 2025-02-11:
The K and I matrices are defined with respect to the reference axis. Airfoils are pinned along their chord according to the non dimensional field `pitch_axis`. This documentation should help https://windio.readthedocs.io/en/latest/source/blade.html
In addition to VABS, you can try the open-source ANBA code, which is called here https://github.com/ptrbortolotti/SONATA/tree/develop/examples/1_IEA15MW
This is how we generated the K and I matrices for the IEA15

vacupenda, 2025-04-09:
Excuse me, I was wondering if you could help clarify something about the material parameters. Could you please explain why several strength parameters are set to zero in this specification? Additionally, since the Poisson's ratio isn't provided, I'm trying to determine the appropriate strength criterion to apply. I would greatly appreciate your insights on the rationale behind these particular parameter configurations.
[image]

closed 2025-04-10 by ptrbortolotti

ptrbortolotti, 2026-02-09:
windIO v2.0 (https://github.com/IEAWindSystems/windIO) ha new documentation that lives here https://ieawindsystems.github.io/windIO/main/index.html

the differences in K and I matrices are likely from differences in the meshes. 

This paper https://doi.org/10.1016/j.compstruct.2020.112755 shows a comparison between ANBA and VABS. For the same mesh K and I matrices should be nearly identical except for some off-diagonal terms caused by pre-twist and pre-curvature of the beam. For a standard wind turbine blade these terms seem negligible (although I have not done a proper check about this)

whhong-research, 2026-02-12:
Thank you very much for your reply — it’s genuinely helpful!

Sorry I didn’t explain my question and testing background clearly at first. My original goal was to use a cross‑section FE tool (such as **VABS** or **BECAS**) to compute strains for a given IEA 15‑MW blade operating condition, and then perform a fatigue assessment based on those strains.

Yes, I obtained access to VABS (academic/free license). During my study I realized that **OpenFAST/BeamDyn** uses stiffness and inertia matrices that were generated via **SONATA** (i.e., using **ANBA**, as you mentioned) as inputs. So my current understanding is:

1. The BeamDyn stiffness/mass matrices in the IEA‑15 dataset were generated with **SONATA calling ANBA**.  
2. In principle, VABS results should be very close to ANBA for the same cross-section, though differences may appear especially in **off‑diagonal coupling terms** (the literature also notes that VABS can account for effects such as initial twist and curvature).  
3. However, when I ran VABS for a specific section, I observed the discrepancies I posted (notably in **GA\_x** and **GJ**, and also a sign change in one coupling term). I suspect this might be *unreasonable*, but since I’m just starting with VABS it’s also quite possible that I configured something incorrectly in my VABS inputs, which then propagated into the comparison.   （Also, apologies for raising this here if it’s not really within the intended scope of this repository/discussion.）

From a workflow perspective, I’m trying to confirm whether the following is a reasonable and commonly used approach:

- Use a cross‑section FE tool (VABS/BECAS/…) to compute $$\mathbf{K}$$ and $$\mathbf{I}$$ (and the strain/stress recovery matrices),  
- use those $$\mathbf{K}$$ and $$\mathbf{I}$$ as **BeamDyn** inputs in **OpenFAST**,  
- run OpenFAST to obtain loads time histories,  
- map the load time histories to strains/stresses at a specific point using the recovery matrices (essentially matrix operations),  
- then use the resulting stress/strain time histories for fatigue analysis.

Does this overall pipeline look correct to you?

Thanks again for your patience. I’m new to this area, so my explanation may still be a bit unclear — sorry about that, and I really appreciate your help.

ptrbortolotti, 2026-02-12:
The pipeline looks correct yes. For a rigorous ANBA vs VABS comparison, I'd recommend generating a single mesh and pass it to the two solvers. SONATA fully supported this until a few years ago (see https://gitlab.lrz.de/HTMWTUM/SONATA), but we don't have access to VABS and so we stopped supporting the portion of the code running VABS