IEAWindSystems/IEA-15-240-RWT #250: Is there manuscript to plot blade lay-up structure?

winder365, 2025-02-28:
Dear all,

Is there manuscript to plot blade lay-up structure like this based on the .yaml file of the IEA-15 MW wind turbine?

[image]

Thank you for your attention.

gbarter, 2025-02-28:
Those plots are auto-generated around [here](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/master/WISDEM/generateTables.py#L432).  If you have WISDEM installed, you can run the [run_model.py](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/master/WISDEM/run_model.py) script to generate the plots.

winder365, 2025-03-06:
Dear @gbarter 

Thank you for you help!

Best wishes,

closed 2025-04-10 by ptrbortolotti