IEAWindSystems/IEA-15-240-RWT #251: Error When Modifying the Sweep Parameters of the 15MW Wind Turbine Blades

vacupenda, 2025-02-28:
## Description
I modified the parameter kp_yr in the file [IEA-15-240-RWT_BeamDyn.dat](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn.dat) to study the impact of sweep magnitude on the wind turbine. However, when I changed the tip sweep to 6 meters, an error occurred. I would like to know whether the problem lies in the modification or if a 6-meter sweep displacement is not allowed. The error messages are as follows:

Running ROSCO-v2.9.0
A wind turbine controller framework for public use in the scientific field
Developed in collaboration: National Renewable Energy Laboratory
                            Delft University of Technology, The Netherlands
------------------------------------------------------------------------------
Generator speed:    7.6 RPM, Pitch angle:   1.0 deg, Power:     0.0 kW, Est. wind Speed:  11.1 m/s

FAST_Solution:FAST_AdvanceStates:SrvD_UpdateStates:DLL_controller_call:BladedInterface option was
designed for an explicit-loose coupling scheme. Using last calculated values from DLL on all
subsequent calls until time is advanced. Warning will not be displayed again.

Warning: SkewedWakeCorrection encountered a large value of chi (-146.16 deg), so the yaw
correction will be limited. This warning will not be repeated though the condition may persist.
See the AD15 chi output channels, and consider turning off the Pitt/Peters skew model (set
SkewMod=1) if this condition persists.

FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 47, blade 1):UA_CalcOutput:UA_BlendSteady:Temporarily turning off UA due to high angle
of attack or low relative velocity. This warning will not be repeated though the condition may
persist.
FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 47, blade 1):UA_CalcOutput:Mach number exceeds 0.3. Theory is invalid. This warning will
not be repeated though the condition may persist.

The BEM solution is being turned off due to low TSR.  (TSR = 1.1806). This warning will not be
repeated though the condition may persist. (See GeomPhi output channel.)

FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_CalcOut
put(node 46, blade 1):UA_CalcOutput:Mach number exceeds 1.0. Equations cannot be evaluated.

 OpenFAST encountered an error at simulation time 4.00000E-03 of 600 seconds.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

vacupenda, 2025-02-28:
> ## Description
> I modified the parameter kp_yr in the file [...]

ptrbortolotti, 2025-02-28:
Hi @vacupenda,
the two 3D axis are different. AeroDyn takes as input the 3D line connecting the aerodynamic centers. BeamDyn takes as input an arbitrary axis, which is however not the line going through the aerodynamic centers, rather the axis that pins each airfoil along span according to this field https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml#L24
When we generate the two files in [WEIS](https://github.com/WISDEM/WEIS), we take the 3D coordinates of the reference axis https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml#L27 and we then find the coordinates of the aerodynamic center using this function https://github.com/WISDEM/WEIS/blob/0da256fe7b02de10cf0726830b3f1da26c087c8f/weis/aeroelasticse/openmdao_openfast.py#L2710
I hope this helps!
Best regards,

vacupenda, 2025-03-03:
Thank you very much for providing the explanation on how to calculate BlCrvAC and BlSwpAC. It calls this function:

`def get_ac_axis(self, inputs):   
  
    # Get the absolute offset between pitch axis (rotation center) and aerodynamic center
    ch_offset = inputs['chord'] * (inputs['ac'] - inputs['le_location'])
    # Rotate it by the twist using the AD15 coordinate system
    x , y = util.rotate(0., 0., 0., ch_offset, -np.deg2rad(inputs['theta']))
    # Apply offset to determine the AC axis
    BlCrvAC = inputs['ref_axis_blade'][:,0] + x
    BlSwpAC = inputs['ref_axis_blade'][:,1] + y
    
    return BlCrvAC, BlSwpAC`
I have some questions about two variables in this function. The first parameter, ac, represents the aerodynamic center of the airfoil, but I’m not sure where to find this data. The second parameter, le_location, is it defined in this file: [IEA-15-240-RWT.yaml#L24](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml#L24)?

The BlCrvAC and BlSwpAC I need to solve are located in the coordinate system of the entire blade. Therefore, when I change the parameters in this file: [IEA-15-240-RWT.yaml#L27](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml#L27), these values will also change.

You mentioned that 'AeroDyn15_blade.dat' and 'BeamDyn.dat' are generated by [WEIS](https://github.com/WISDEM/WEIS). What steps should I take to generate these files? What programs do I need to install and run?

ptrbortolotti, 2025-03-03:
Hi,
The aerodynamic center line can be built from the yaml file by combining the spanwise positions of the airfoils (https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml#L17) and each aerodynamic center per airfoil (https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/6b3fd8ac92935f7f6fe1810a909afc2430f6896f/WT_Ontology/IEA-15-240-RWT.yaml)
You are correct about #2
The installation instructions for WEIS are available here https://github.com/WISDEM/WEIS. You should be able to install it on Windows (readme is slightly outdated, thanks for reminding us to update it)
Note that WEIS does not yet support BeamDyn. To generate those files, you need the tool SONATA, which comes with an example for the IEA15, see here https://github.com/ptrbortolotti/SONATA/tree/develop/examples/1_IEA15MW
A long term goal is to bring WEIS and SONATA together
And lastly yes, to introduce sweep, you can simply add a `y` component to the `ref_axis` field and regenerate the openFAST model with WEIS (for AD15) and SONATA (for BeamDyn)

vacupenda, 2025-03-04:
When performing calculations using the newly generated files IEA-15-240-RWT_AeroDyn15_blade.dat and IEA-15-240-RWT_BeamDyn.dat, the error message shown in the figure appears. Could you please tell me what the issue is and how it can be resolved?

[image]

gbarter, 2025-03-04:
It looks like your simulation is going unstable, likely from too high of a time step.  I would suggest lowering your time step.  You may need to iterate with trial-and-error to find a value that is consistently stable.

vacupenda, 2025-03-06:
Another issue I encountered is that while the current calculations are functioning properly, the wind speed I set was 10.59 m/s, but during the calculation, it changed to the value shown in the diagram. I’m not sure where the problem lies.
[image]

closed 2025-04-10 by ptrbortolotti