IEAWindSystems/IEA-15-240-RWT #252: Definition of the fiber direction of the blade layup in .yaml and SONATA

winder365, 2025-03-06:
Dear @ptrbortolotti 

I am using the SONATA to calculate the section property of wind turbine blade, namely the 6*6 stiffness and mass matrixes.

I want to change the fiber orientation of Carbon UD used in spar cap. I have known that from 'windIO’s documentation' that 'For composite laminates, the orientation of the fibers in radians can be specified. Looking from blade root, positive angles represent a rotation of the fibers towards the leading edge of the blade. Note that the angles are in respect to the cross section local reference system, not the reference system at blade root.' (https://windio.readthedocs.io/en/latest/source/blade.html#internal-structure-2d-fem)

So,  if I want to change the fiber orientation of Carbon UD to 15 deg ( namely the theta_ply in the following picture is 15 deg), is it correct to changing the fiber_orientation value of the Spar_Cap_PS and Spar_Cap_PS from 0 to 0.2618 (in rad, namely 15deg)  in .yaml file?
 For example, from
` name: Spar_Cap_PS
  material: CarbonUD
fiber_orientation:
     grid: [0.0, 1.0]
      values: [0.0, 0.0]`
to
` name: Spar_Cap_PS
  material: CarbonUD
fiber_orientation:
     grid: [0.0, 1.0]
      values: [0.2618, 0.2618]`

[image]

ptrbortolotti, 2025-03-06:
hi @winder365, your interpretation is correct. 

note however that @JustinPorter88 has recently been working on [SONATA](https://github.com/WISDEM/SONATA), and he's concerned that SONATA might no longer be applying this angle correctly once elements are split from quads into triangles (this split is required by the solver [anba4](https://github.com/ANBA4/anba4)).

I've never used the fiber angle in any of my projects, so maybe do some vetting before proceeding. @JustinPorter88 please provide guidance if you can

also a curiosity on my end: these days CFRP is made of pultruded planks. @winder365, can you have planks with fibers oriented at 15 degrees?

JustinPorter88, 2025-03-06:
I'll dig into the code of SONATA and double check what it is doing/if we have any verification examples.

winder365, 2025-03-09:
Dear @ptrbortolotti and @JustinPorter88 

Thank you for your efficient response for this issue. I will have a try according to your suggestion! As soon as I have any new findings, I will reply here.

By the way, is this issue related to the 'elements are split from quads into triangles in ANBA4' as @ptrbortolotti mentioned above?

JustinPorter88, 2025-03-10:
It does not appear to be related to elements getting split into triangles. That operation directly copies over the two rotation angles, so should not be a problem. 

Rather, the issue appears to be that these two rotation angles just get directly applied while the WindIO documentation implies that one would need to check if an element is on the suction or pressure side and then flip the sign of the fiber angle rotation relative to the outward (or inward) normal of the airfoil cross-section.

There are some edge cases where SONATA directly creates triangles instead of quads (e.g., around sharp trailing edges). In those cases, the element orientation `theta_11` may sometimes be wrong. You can try plotting that field to verify that it looks good. It should probably be the attribute `theta_11` or `theta_1`.