IEAWindSystems/IEA-15-240-RWT #253: IEA-15-240-RWT-UMaineSemi

mocominnie, 2025-03-13:
Hi, 

I'm trying to run the IEA-15-240-RWT-UMaineSemi in the latest version of openfast but I get this error, 

FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid numerical input for file
 "IEA-15-240-RWT-UMaineSemi.fst" occurred while trying to read MHK.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

I'm not entirely sure what the cause is or what I would need to do to get it working.

Thanks for your help

ptrbortolotti, 2025-03-13:
This is probably a mismatch of OpenFAST versions. The develop branch works for OpenFAST v4.0.x, whereas the master branch of this repo is still at v3.5.x. Our apologies, we have not updated master in a while. We will do that soon. Before that, please confirm that this is your issue

ptrbortolotti, 2025-03-13:
See https://github.com/IEAWindSystems/IEA-15-240-RWT/pull/246

closed 2025-04-10 by ptrbortolotti