IEAWindSystems/IEA-15-240-RWT #255: IEA-15-240-RWT-Monopile

BULL-Paddywack, 2025-03-24:
Hi,

I'm trying to run the IEA-15-240-RWT-Monopile in the latest version of fastfarm but I get this error,

Farm_Initialize:T1:Farm_InitFAST:FWrap_Init:MeshMapCreate:CreateLoadMap_L2_to_L2:CreateMapping_Pro
jectToLine2:Node 19 does not project onto any line2 element. Closest distance is 1.9137 m.

Aborting FAST.Farm.

I have tried to change some data in IEA-15-240-RWT-Monopile_AeroDyn15.dat and IEA-15-240-RWT-Monopile_ElastoDyn_tower.dat but it doesn't work.

I'm not entirely sure what the cause is or what I would need to do to get it working.

Thanks for your help!

gbarter, 2025-03-24:
Which version of OpenFAST are you using?  The OpenFAST v4 files were just stamped as a release at the end of last week.

BULL-Paddywack, 2025-03-25:
I tried it with openfast 4.0.0 and 3.4.0, and the problem occurred in both versions.

BULL-Paddywack, 2025-03-25:
Because I wanted to reproduce the results of an article using version 3.4.0, I downloaded the code myself as a zip. I still want to ask where I should make modifications to solve this problem, thanks a lot.

gbarter, 2025-03-25:
If you really, really want to use OpenFAST v3.4, the [release notes suggest](https://github.com/IEAWindSystems/IEA-15-240-RWT/releases/tag/v1.1.4) you'd have to go back to v1.1.4 in this repository from Apr 2023.  There have been bug fixes and improvements since then, separate from the OpenFAST API changes, but you'd have to make use of the Github history to bring those in.

lu8lu, 2025-04-07:
Hello, I have the same problem. Have you solved it? Did you modify any parameters?

gbarter, 2025-04-07:
> Hello, I have the same problem. Have you solved it? Did you modify any parameters?

What version of OpenFAST are you using?

BULL-Paddywack, 2025-04-08:
> Hello, I have the same problem. Have you solved it? Did you modify any parameters?

Haven't solved it yet. I'm now using the v4.0.2. I have changed the hubheight and diameter in .fstf and quoted the IEA-15-240-RWT-Monopile.fst in LESinflow in reg-tests. I think it's the problem of CreateMapping of the tower but I haven't got the idea about how to solve it.

gbarter, 2025-04-08:
In talking to the OpenFAST team, there are some known bugs in FAST.Farm with OpenFAST 4.0.2.  A fix in 4.0.3 should be released any day.

mayankchetan, 2025-04-08:
Yes, they files were updated for 4.0.0, and no API changes between 4.0.0 & 4.0.2. Running locally I'm not able to recreate this issue with just OpenFAST (suspecting its a OpenFAST level error not FAST.Farm).

lu8lu, 2025-04-09:
> > Hello, I have the same problem. Have you solved it? Did you modify any parameters?你好，我也有同样的问题。你解决了吗？你修改了任何参数吗？
> 
> What version of OpenFAST are you using?你使用的是哪个版本的 OpenFAST？

I first used Turbsim to generate flow field data

         25   NumGrid_Z       - Vertical grid-point matrix dimension
         160   NumGrid_Y       - Horizontal grid-point matrix dimension
       0.01   TimeStep        - Time step [seconds]
        1100   AnalysisTime    - Length of analysis time series [seconds] (program will add time if necessary: AnalysisTime = MAX(AnalysisTime, UsableTime+GridWidth/MeanHHWS) )
"ALL"         UsableTime      - Usable length of output time series [seconds] (program will add GridWidth/MeanHHWS seconds unless UsableTime is "ALL")
         95   HubHt           - Hub height [m] (should be > 0.5*GridHeight)
         180   GridHeight      - Grid height [m]
         1600   GridWidth       - Grid width [m] (should be >= 2*(RotorRadius+ShaftLength))

Then set up a single fan in .fstf and test to get the wake field data

--- AMBIENT WIND: INFLOWWIND MODULE --- [used only for Mod_AmbWind=2 or 3]
1.0                DT_Low             - Time step for low -resolution wind data interpolation; will be used as the global FAST.Farm time step (s) [>0.0]
0.1                DT_High            - Time step for high-resolution wind data interpolation (s) [>0.0]
240                NX_Low             - Number  of low -resolution spatial nodes in X direction for wind data interpolation (-) [>=2]
80                 NY_Low             - Number  of low -resolution spatial nodes in Y direction for wind data interpolation (-) [>=2]
18                 NZ_Low             - Number  of low -resolution spatial nodes in Z direction for wind data interpolation (-) [>=2]
0               X0_Low             - Origin  of low -resolution spatial nodes in X direction for wind data interpolation (m)
-640                  Y0_Low             - Origin  of low -resolution spatial nodes in Y direction for wind data interpolation (m)
5.0                Z0_Low             - Origin  of low -resolution spatial nodes in Z direction for wind data interpolation (m)
10.0              dX_Low             - Spacing of low -resolution spatial nodes in X direction for wind data interpolation (m) [>0.0]
10.0               dY_Low             - Spacing of low -resolution spatial nodes in Y direction for wind data interpolation (m) [>0.0]
10.0               dZ_Low             - Spacing of low -resolution spatial nodes in Z direction for wind data interpolation (m) [>0.0]
112                 NX_High            - Number  of high-resolution spatial nodes in X direction for wind data interpolation (-) [>=2]
32                 NY_High            - Number  of high-resolution spatial nodes in Y direction for wind data interpolation (-) [>=2]
17                 NZ_High            - Number  of high-resolution spatial nodes in Z direction for wind data interpolation (-) [>=2]
"IEA-3.4-130-RWT_InflowWind_steady.dat"           InflowFile         - Name of file containing InflowWind module input parameters (quoted string)
--- WIND TURBINES ---
1                  NumTurbines        - Number of wind turbines (-) [>=1]                         [last 6 columns below used only for Mod_AmbWind=2 or 3]
WT_X    WT_Y   WT_Z   WT_FASTInFile     X0_High  Y0_High  Z0_High  dX_High  dY_High  dZ_High
(m)     (m)    (m)    (string)          (m)      (m)      (m)      (m)      (m)      (m)
500.0   0.0    10.0    "WT2_5.fst"   340.0   -160.0   10.0      10.0    10.0     10.0

First the problem

FAST.Farm-v3.4.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 2021
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jan 13 2023
 - Time: 11:00:06
Execution Info:
 - Date: 04/07/2025
 - Time: 14:54:07+0800

 Heading of the FAST.Farm input file:
   FAST.Farm input file, using two turbines separated by 3D downstream and an offset of 30m,
   turbulent inflow, yaw maneuver for WT1
Running AWAE.
Running InflowWind.

   Reading a 80x25 grid (800 m wide, 5 m to 185 m above ground) with a characteristic wind speed
   of 10.7 m/s. This full-field file was generated by TurbSim (from OpenFAST-v3.4.0) on
   06-Apr-2025 at 22:40:12.

   Processed 110000 time steps of 100-Hz full-field data (period of 1100 seconds).
Running WakeDynamics.
Running FASTWrapper.
Running OpenFAST.
OpenFAST input file heading:
    Generated with AeroElasticSE FAST driver

[... 39 lines cut ...]

   Processed 110000 time steps of 100-Hz full-field data (period of 1100 seconds).
Running WakeDynamics.
Running FASTWrapper.
Running OpenFAST.
OpenFAST input file heading:
    Generated with AeroElasticSE FAST driver

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
 Time: 0 of 600 seconds.

T1:FARM_InitialCO:FWrap_t0:FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SolveOption2
c_Inp2AD_SrvD:InflowWind_CalcOutput:CalcOutput:IfW_4Dext_CalcOutput [position=(-3.262, 2.574,
160.12) in wind-file coordinates]:Interp4D:Outside the grid bounds.

 Aborting FAST.Farm.

I want to know what is the cause of this? What settings in .fstf should I modify to solve the above problem? I tried to regenerate the inflow wind data, but it doesn't seem to be caused by this problem.
Due to version reasons, I'd better use version v3.4.0. I need to solve this problem on this version. How should this be solved?

andrew-platt, 2025-04-09:
@lu8lu, this appears to be an issue with your setup of _FAST.Farm_.  What `Mod_AmbWind` value are you using, and what turbine position?

BULL-Paddywack, 2025-05-05:
@lu8lu, I have already solved the problem, refer to https://forums.nrel.gov/t/fast-farm-failed-load-mapping/2501