IEAWindSystems/IEA-15-240-RWT #256: Question about the additional damping matrix of UMaine platform defined in the wamit input file  IEA-15-240-RWT_2nd.frc

getChaos22, 2025-03-28:
Hi all, 

In the second order wamit file ([IEA-15-240-RWT_2nd.frc](https://github.com/IEAWindSystems/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData/wamit_inputs_2ndOrder/IEA-15-240-RWT_2nd.frc)), an additional damping matrix is defined. There are two questions about this matrix that I would like to seek your help. One is the origin of this damping matrix, or do you have some references on how the damping matrix is obtained? In the [definition report of IEA15MW](https://www.nrel.gov/docs/fy20osti/76773.pdf), only a quadratic drag matrix is given. The other one is the units of this additional damping matrix.  Are the coefficients in the matrix normalized?

Best regards,
Chao Li

luwang00, 2025-04-07:
Hi @getChaos22, if I remember correctly, the linear damping matrix was obtained by assuming 5% critical damping in each of the six platform degrees of freedom. 

The second-order wave excitation depends on the first-order motion RAOs over the incident wave frequencies. On the other hand, the added damping primarily impacts the RAOs near the resonance frequencies, which, for the VolturnUS-S, are usually below the wave frequencies. Above 0.05 Hz, the damping has almost no effect on the surge and pitch RAOs and only limited effect on the heave RAOs (comparing 5% vs 10% critical damping for example). Therefore, the exact choice of damping is not as important for this design _for the purpose of computing second-order potential-flow wave excitation_. 

As to your second question, I believe the damping coefficients are dimensional and not normalized, but you should NOT be using them for any purpose other than getting the second-order potential-flow wave excitation. If you want to obtain accurate resonance response in time-domain simulations, you need better tuned damping/drag coefficients.

getChaos22, 2025-04-08:
Hi @luwang00,

Thanks for your help. Your explanation is quite clear. 

Best regards,
Chao Li

closed 2025-04-08 by getChaos22